jetperch/pyjoulescope_ui #184: ModuleNotFoundError: No module named 'pyjoulescope_driver'

bauerjj, 2023-05-03:
### Joulescope model

JS220

### UI version

1.0.9

### What OS are you seeing the problem on?

Linux Ubuntu amd64 (PC)

### What happened?

Unable to install the joulescope driver. Please see below

```
$ pip3 install joulescope

Collecting pyjls>=0.6.2 (from joulescope)
  Downloading pyjls-0.6.2.tar.gz (270 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 270.1/270.1 kB 43.6 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pyjoulescope-driver>=1.3.7 (from joulescope)
  Using cached pyjoulescope_driver-1.3.7.tar.gz (2.7 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [15 lines of output]
      Traceback (most recent call last):
        File "/home/xxxx/venv/joulescope/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/xxxx/venv/joulescope/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/xxxx/venv/joulescope/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-6h48ccdc/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 341, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
        File "/tmp/pip-build-env-6h48ccdc/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 323, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-6h48ccdc/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 338, in run_setup
          exec(code, locals())
        File "<string>", line 34, in <module>
      ModuleNotFoundError: No module named 'pyjoulescope_driver'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error
```

I tried the same with `pip install pyjoulescope-driver` but same error as above

I also tried building from source and followed this page https://pypi.org/project/pyjoulescope-driver/

```
cd {your/repos/joulescope_driver}
mkdir build && cd build
cmake ..
cmake --build .
```

```
$ cd src/build
$ cmake ..
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:78 (SET_FILENAME):
  Unknown CMake command "SET_FILENAME".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.22)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/xxxxx/Downloads/pyjoulescope_driver-1.3.7/src/build/CMakeFiles/CMakeOutput.log".
```

### What was expected?

joulescope-driver should be installed successfully 

### How to reproduce?

1. Install fresh Ubuntu 22.04 image using python3.10 
2. pip3 install joulescope
3. Or attempt building from source

### Extra information

ubuntu version:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.2 LTS
Release:	22.04
Codename:	jammy
```

python version:

```
$ python3 --version
Python 3.10.6
```

mliberty1, 2023-05-03:
Hi @bauerjj and thanks for reporting this issue.  I confirm that `python3 -m pip install pyjoulescope_driver` gives the same error with a clean venv on Ubuntu 22.04 LTS.  That error is entirely unhelpful, so I will have to investigate.  Python packaging recently obsoleted the python setup.py` method, so I suspect that this is related.

In the meantime, here is a process to build and install from source:

```
python3 -m venv ~/venv/tmp
. ~/venv/tmp/bin/activate
pip3 install Cython numpy requests setuptools wheel
cd {path/to}/joulescope_driver
python3 setup.py bdist_wheel
pip3 install dist/pyjoulescope_driver*
```

Does this work for you for now?

mliberty1, 2023-05-17:
Fixed in UI 1.0.15 with pyjoulescope_driver 1.3.9.

closed 2023-05-17 by mliberty1