jetperch/pyjoulescope_ui #193: qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.

immanuelplattner, 2023-06-15:
### Joulescope model

JS220

### UI version

1.0.22

### What OS are you seeing the problem on?

Linux Ubuntu amd64 (PC)

### What happened?

I tried to start the *joulescope_ui* from the terminal (on Ubuntu 22.04.02 LTS), but it did not start. Instead it just gave the error message `qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.`. Below it the terminal session extract:

```
$ python3 -m joulescope_ui
Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.22       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.22"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-43-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/usr/bin/python3",
    "frozen": false
  }
}

=====
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: wayland-egl, wayland, eglfs, minimal, xcb, vkkhrdisplay, minimalegl, linuxfb, offscreen, vnc.

Aborted (core dumped)
$ joulescope ui
Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.19       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.19"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-43-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/usr/bin/python3",
    "frozen": false
  }
}

=====
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: wayland-egl, wayland, eglfs, minimal, xcb, vkkhrdisplay, minimalegl, linuxfb, offscreen, vnc.

Aborted (core dumped)
~/joulescope_1_0_22$ ./joulescope_launcher 
Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.22       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.22"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.11.3 (main, Apr  6 2023, 07:55:46) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-43-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/home/plat/joulescope_1_0_22/joulescope_launcher",
    "frozen": "/home/plat/joulescope_1_0_22"
  }
}

=====
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: wayland-egl, wayland, eglfs, minimal, xcb, vkkhrdisplay, minimalegl, linuxfb, offscreen, vnc.

Aborted (core dumped)
```

### What was expected?

UI should have opened.

### How to reproduce?

Try and start the *joulescope_ui* from the terminal.

### Extra information

I installed *joulescope_ui* version 1.0.22 on my computer running Ubuntu 22.04.2 LTS. I used the method described in the *readme.md* section *Install using pypi*. At first there was an error, because the package *PySide6* was not found (see terminal session extract below).

```
$ python3 -m pip install -U --upgrade-strategy=eager joulescope_ui
...
...
...
Collecting packaging (from QtPy->joulescope_ui)
  Downloading packaging-23.1-py3-none-any.whl (48 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 48.9/48.9 kB 11.1 MB/s eta 0:00:00
Building wheels for collected packages: joulescope_ui, pyperclip
  Building wheel for joulescope_ui (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [651 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib
      creating build/lib/joulescope_ui
      copying joulescope_ui/main.py -> build/lib/joulescope_ui
      copying joulescope_ui/pubsub.py -> build/lib/joulescope_ui
      copying joulescope_ui/profile.py -> build/lib/joulescope_ui
      copying joulescope_ui/metadata.py -> build/lib/joulescope_ui
      ...
      ...
      ...
      running install_egg_info
      Copying joulescope_ui.egg-info to build/bdist.linux-x86_64/wheel/joulescope_ui-1.0.19.egg-info
      running install_scripts
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-uddxnt5y/joulescope-ui_9325dbde083c48008d25d0138efb7735/setup.py", line 145, in <module>
          setuptools.setup(
        File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 153, in setup
          return distutils.core.setup(**attrs)
        File "/usr/lib/python3.10/distutils/core.py", line 148, in setup
          dist.run_commands()
        File "/usr/lib/python3.10/distutils/dist.py", line 966, in run_commands
          self.run_command(cmd)
        File "/usr/lib/python3.10/distutils/dist.py", line 985, in run_command
          cmd_obj.run()
        File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 335, in run
          self.run_command('install')
        File "/usr/lib/python3.10/distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/usr/lib/python3.10/distutils/dist.py", line 985, in run_command
          cmd_obj.run()
        File "/tmp/pip-install-uddxnt5y/joulescope-ui_9325dbde083c48008d25d0138efb7735/setup.py", line 124, in run
          convert_qt_ui()
        File "/tmp/pip-install-uddxnt5y/joulescope-ui_9325dbde083c48008d25d0138efb7735/setup.py", line 59, in convert_qt_ui
          rcc_path = qt_rcc_path()
        File "/tmp/pip-install-uddxnt5y/joulescope-ui_9325dbde083c48008d25d0138efb7735/setup.py", line 43, in qt_rcc_path
          import PySide6
      ModuleNotFoundError: No module named 'PySide6'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for joulescope_ui
  Running setup.py clean for joulescope_ui
  Building wheel for pyperclip (setup.py) ... done
  Created wheel for pyperclip: filename=pyperclip-1.8.2-py3-none-any.whl size=11137 sha256=0a42c648d3d2cf505bcd5af405c8e9608d2d1c9ea3b31574be589f649fc65f1a
  Stored in directory: /home/plat/.cache/pip/wheels/04/24/fe/140a94a7f1036003ede94579e6b4227fe96c840c6f4dcbe307
Successfully built pyperclip
Failed to build joulescope_ui
```

So I installed the missing package *PySide6* using *pip* and ran the installation again. This time there was no error.

mliberty1, 2023-06-15:
Does installing Qt6 help?

    sudo apt install libudev-dev qt6-base-dev qt6-3d-dev

immanuelplattner, 2023-06-16:
@mliberty1 Thanks for the fast answer. Unfortunately, that does not help. I still have the same error. Also, when I execute the *joulescope_ui* I installed from pypi the version is `"ui_version": "1.0.19"`. I did not notice that yesterday, but it was already the same version. Is the package on pypi not the newest version? Because the "readme.md* says so in section *Installing using pypi*:

> If you just want to run the latest released version of the UI, use this option!

Here is the terminal session from trying to execute the *joulescope_ui" package installed from pypi:

```
~$ python3 -m joulescope_ui
Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.19       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.19"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-43-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/usr/bin/python3",
    "frozen": false
  }
}

=====
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: wayland-egl, wayland, eglfs, minimal, xcb, vkkhrdisplay, minimalegl, linuxfb, offscreen, vnc.

Aborted (core dumped)
```

I also downloaded the newest version (which is 1.0.22 ) for Ubuntu from the [website](https://www.joulescope.com/download) and tried to execute that. Still no luck, as you can see below.
```
~$ cd joulescope_1_0_22/
~/joulescope_1_0_22$ ./joulescope_launcher 
Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.22       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.22"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.11.3 (main, Apr  6 2023, 07:55:46) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-43-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/home/plat/joulescope_1_0_22/joulescope_launcher",
    "frozen": "/home/plat/joulescope_1_0_22"
  }
}

=====
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: wayland-egl, wayland, eglfs, minimal, xcb, vkkhrdisplay, minimalegl, linuxfb, offscreen, vnc.

Aborted (core dumped)
~/joulescope_1_0_22$ python3 -m joulescope_ui 
Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.22       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.22"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-43-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/usr/bin/python3",
    "frozen": false
  }
}

=====
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: wayland-egl, wayland, eglfs, minimal, xcb, vkkhrdisplay, minimalegl, linuxfb, offscreen, vnc.

Aborted (core dumped)
```

The first version I downloaded from the website is 0.10.13 and that is still running just fine on my system, although there are two warnings that occur over and over and over as you can see int the terminal session below.

```
~/joulescope_1_0_22$ cd ../joulescope_0_10_13/
~/joulescope_0_10_13$ ./joulescope_launcher 
*****************************
* Joulescope User Interface *
* UI Version = 0.10.13      *
* Driver Version = 1.0.15   *
*****************************

info = {
  "joulescope": {
    "ui_version": "0.10.13",
    "driver_version": "1.0.15"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.10.6 (main, Nov 14 2022, 16:10:14) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-43-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/home/plat/joulescope_0_10_13/joulescope_launcher",
    "frozen": "/home/plat/joulescope_0_10_13",
    "paths": {
      "dirs": {
        "app_path": "/home/plat/.joulescope",
        "config": "/home/plat/.joulescope/config",
        "log": "/home/plat/.joulescope/log",
        "firmware": "/home/plat/.joulescope/firmware",
        "themes": "/home/plat/.joulescope/themes",
        "data": "/home/plat/Documents/joulescope",
        "update": "/home/plat/.joulescope/update"
      },
      "files": {
        "config": "/home/plat/.joulescope/config/joulescope_config.json"
      }
    }
  }
}

=====
WARNING:2023-06-16 09:21:41,350:js220_usb.c:1353:jsdrv:stream in ignore on inactive port 21
WARNING:2023-06-16 09:21:41,350:js220_usb.c:1353:jsdrv:stream in ignore on inactive port 22
...
...
...
```

Thanks for the support.

mliberty1, 2023-06-16:
I just published 1.0.22 to pypi.  Sorry about that!  We completely updated the build system so that everything is built using GitHub Actions, but we left out the publish to pypi step.  It will be fixed for future releases.  You should be able to get 1.0.22 now:

    python3 - m pip install -U --upgrade-strategy eager joulescope_ui

However, I doubt that this will fix the issue you are seeing.

A [Google search](https://www.google.com/search?q=t.qpa.plugin%3A+Could+not+load+the+Qt+platform+plugin+%22xcb%22+in+%22%22+even+though+it+was+found.) shows that this is not an uncommon problem.  It also seems that there are lots of different solutions.  The simplest possible solution seems to be:

    sudo apt install libxcb-xinerama0 

Does this fix it?

If not, is there something different about your system that might cause a difference in the graphics rendering?  We build and test on stock Ubuntu 22.04 LTS.

immanuelplattner, 2023-06-16:
@mliberty1 Thank you for publishing the newest version to pypi. I installed the new package using  `python3 -m pip install -U --upgrade-strategy=eager joulescope_ui` and had no issues during the installation. To nobody's surprise, this did not fix the issue.

I was aware that lots of people had a similar issue before. But I saw that the origin was from a lot of different softwares and I just assumed that their solution would not work for me. I tried `sudo apt install libxcb-xinerama0` but it was already installed. So I just started looking through the Google search provided by @mliberty1. The [second search result](https://intellij-support.jetbrains.com/hc/en-us/community/posts/11003640644242-Ubuntu-20-04-xcb) I looked at contained the solution, which was to install the missing library *libxcb-cursor0* using `sudo apt install libxcb-cursor0`.

Now both versions run, the package installed using pypi as well as the one I downloaded from your website (see terminal sessions below).

```
~$ joulescope ui
Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.22       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.22"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-43-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/usr/bin/python3",
    "frozen": false
  }
}

=====
qt.dbus.integration: Could not connect "org.freedesktop.IBus" to globalEngineChanged(QString)
WARNING:2023-06-16 14:48:34,620:pubsub.py:1341:joulescope_ui.pubsub:Could not unregister <class 'joulescope_ui.widgets.debug.debug_widget.DebugWidget'> - invalid spec
WARNING:2023-06-16 14:48:34,765:waveform_widget.py:551:joulescope_ui.widgets.waveform.waveform_widget:No default source available
WARNING:2023-06-16 14:48:35,053:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-000185/sources/1/info
WARNING:2023-06-16 14:48:35,329:backend.c:289:jsdrv:bulk out returned 5 NO DEVICE
WARNING:2023-06-16 14:48:35,330:jsdrv.c:763:jsdrv:API command u/js220/000185/@/!close invoked on jsdrv thread with timeout.  Forcing timeout=0.
WARNING:2023-06-16 14:48:36,593:js220_usb.c:503:jsdrv:d_ctrl_req(2) returned 8
WARNING:2023-06-16 14:48:36,594:jsdrv.c:763:jsdrv:API command u/&js220/000185/@/!close invoked on jsdrv thread with timeout.  Forcing timeout=0.
WARNING:2023-06-16 14:48:36,594:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-000185-UPDATER/actions/!finalize
WARNING:2023-06-16 14:48:38,858:js220_usb.c:503:jsdrv:d_ctrl_req(2) returned 8
WARNING:2023-06-16 14:48:38,860:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-000185-UPDATER/actions/!finalize
WARNING:2023-06-16 14:48:38,864:jsdrv.c:763:jsdrv:API command u/&js220/000185/@/!close invoked on jsdrv thread with timeout.  Forcing timeout=0.
WARNING:2023-06-16 14:48:41,053:backend.c:253:jsdrv:libusb_submit_transfer returned -4
WARNING:2023-06-16 14:48:41,056:js220_usb.c:503:jsdrv:d_ctrl_req(2) returned 8
WARNING:2023-06-16 14:48:41,056:jsdrv.c:763:jsdrv:API command u/&js220/000185/@/!close invoked on jsdrv thread with timeout.  Forcing timeout=0.
WARNING:2023-06-16 14:48:42,056:js220_usb.c:330:jsdrv:ll_await timed out
Exception in thread Thread-5 (_run):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
WARNING:2023-06-16 14:48:42,057:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-000185-UPDATER/actions/!finalize
    self._target(*self._args, **self._kwargs)
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/devices/jsdrv/js220_updater.py", line 309, in _run
    self._driver.close(self.device_path)
  File "pyjoulescope_driver/binding.pyx", line 877, in pyjoulescope_driver.binding.Driver.close
  File "pyjoulescope_driver/binding.pyx", line 591, in pyjoulescope_driver.binding._handle_rc
TimeoutError: jsdrv_close timed out | u/&js220/000185
WARNING:2023-06-16 14:48:47,370:backend.c:253:jsdrv:libusb_submit_transfer returned -4
WARNING:2023-06-16 14:48:47,370:js220_usb.c:503:jsdrv:d_ctrl_req(2) returned 8
WARNING:2023-06-16 14:48:47,370:jsdrv.c:763:jsdrv:API command u/&js220/000185/@/!close invoked on jsdrv thread with timeout.  Forcing timeout=0.
WARNING:2023-06-16 14:48:47,371:jsdrv.c:386:jsdrv:device_lookup(u/&js220/000185/@/!close#) => u/&js220/000185 failed
WARNING:2023-06-16 14:48:47,371:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-000185-UPDATER/actions/!finalize
WARNING:2023-06-16 14:48:47,371:jsdrv.c:615:jsdrv:no device match for u/&js220/000185/@/!close#
WARNING:2023-06-16 14:48:47,923:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-000185/sources/1/info
WARNING:2023-06-16 14:49:34,525:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JsdrvStreamBuffer:001/settings/signals/JS220-000185.v/range
WARNING:2023-06-16 14:49:34,525:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JsdrvStreamBuffer:001/settings/signals/JS220-000185.p/range
WARNING:2023-06-16 14:49:34,525:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JsdrvStreamBuffer:001/settings/signals/JS220-000185.v/range
[... cut ...]
libxcb-xfixes0/jammy,now 1.14-3ubuntu3 i386 [installed,automatic]
libxcb-xinerama0/jammy,now 1.14-3ubuntu3 amd64 [installed]
libxcb-xinput0/jammy,now 1.14-3ubuntu3 amd64 [installed,automatic]
libxcb-xkb1/jammy,now 1.14-3ubuntu3 amd64 [installed]
libxcb-xv0/jammy,now 1.14-3ubuntu3 amd64 [installed,automatic]
libxcb1-dev/jammy,now 1.14-3ubuntu3 amd64 [installed,automatic]
libxcb1/jammy,now 1.14-3ubuntu3 amd64 [installed]
libxcb1/jammy,now 1.14-3ubuntu3 i386 [installed,automatic]
```

Thanks for all your effort.

mliberty1, 2023-06-16:
Excellent!  Thank you for your work in tracking down this issue.  We have already updated our installation instructions.

For future people stumbling on this issue, the process is to first find the missing libraries:

    export QT_DEBUG_PLUGINS=1
    python3 -m joulescope_ui

In this case, it was missing libxcb-cursor0, which can be fixed by:

    sudo apt install libxcb-cursor0

According to Qt, here are the required [dependencies](https://doc.qt.io/qt-6/linux-requirements.html).

closed 2023-06-16 by mliberty1