jetperch/pyjoulescope_ui #195: Trigger on waveform for automated testing

FrankZyyz, 2023-06-27:
### Joulescope model

JS220

### UI version

1.0.22

### Your idea

JS220 has OUT0 and OUT1, now it only can be set by mouse click the joulescope_ui button.
I would like OUT0 can output a waveform like HIGH for 200ms, LOW for 1000ms, repeat for 1000 times, 
Above function can help to setup some automated testing with my device.(battery test or something else).

In fact, I already saw this in features_future.md.

I would like to get some hints about how to adjust joulescope_ui code and call some module to create this waveform.
(Like create a thread and set the timing, control the OUT0/OUT1 status.)
Thanks.

mliberty1, 2023-06-27:
Hi @FrankZyyz , and thank you for the feature request.  I changed the title to hopefully clarify that you want to use this signal to trigger external equipment, not the JS220 itself.  Did I understand correctly?

The feature to generate digital waveforms for OUT0, OUT1, and Trigger OUT is definitely on our list.  We will implement this int the JS220 FPGA to give microsecond accuracy and precision.  However, we have not figured out the details of the feature.  What you described is a [PWM](https://en.wikipedia.org/wiki/Pulse-width_modulation) waveform, which it will certainly support.  We do not have a timeframe for this feature yet, unfortunately.

While you could implement this using host-side python, the precision will not be great.  Getting a Windows, macOS, or Linux computer to do anything with millisecond precision is a losing battle.

In the meantime, perhaps you can use a Raspberry PI Pico to generate this waveform?  Here is the [quick PWM reference](https://docs.micropython.org/en/latest/rp2/quickref.html#pwm-pulse-width-modulation).  You can also check out the [JS220 Evaluation Kit 1 firmware](https://github.com/jetperch/js220_evk1/blob/main/firmware/evk.py).  It starts with a comment about the ways to control digital signals:

```
MicroPython with the Raspberry PI RP2040 microcontroller has five different
ways to toggle signals over time:
1. GPIO set from REPL
2. GPIO set from for/while loop with time.sleep()
3. GPIO set from machine.Timer ISR
4. PWM
5. PIO
```

What do you think?

FrankZyyz, 2023-06-28:
@mliberty1  Thank you for your reply.
Your understand is correct about changed title.

And thanks for your warn about OS precision and MicroPython solution.

mliberty1, 2024-04-30:
Partially done.  1.1.6 now has the Trigger widget, which you can use to detection conditions and take actions, such as drive a GPO.  However, it does not yet have the ability to generate a PWM signal using the JS220 FPGA.