jetperch/pyjoulescope_ui #197: Crash when waking up from sleep

romain145, 2023-06-27:
### Joulescope model

JS110

### UI version

1.0.22

### What OS are you seeing the problem on?

Windows 10

### What happened?

When the laptop wakes up from sleep, Joulescope crashes.
When Joulescope is restarted, the JS110 is not detected.

### What was expected?

UI should not crash.
JS110 should be detected when the application is launched.

### How to reproduce?

1. ...
2. ...
3. ...

mliberty1, 2023-06-27:
Hi @romain145 - Would you be able to share the Joulescope UI log file for when this happened?  You should be able to duplicate this, and then use Windows File Explorer to open %LOCALAPPDATA%\joulescope\log.  You can then post the most recent log file or email it to support _at_ joulescope _dot_ com.  If you email it, mention pyjoulescope_ui #197 in the subject line.

I quickly attempted to duplicate this on my Win 11 laptop, but I could not.  I ran  Joulescope UI 1.0.22 with a JS110 connected. 
 While in Waveform View, I closed the lid.  Upon reopening, it kept working.  While I ensured that "sleep" was enabled, I know that I have changed the default power settings.

romain145, 2023-06-28:
This is the log file
[joulescope_20230627_191242_28820.log](https://github.com/jetperch/pyjoulescope_ui/files/11893683/joulescope_20230627_191242_28820.log)

mliberty1, 2023-06-28:
Thank you for the log file, @romain145 .  That log file shows that the JS110 communication does not come back immediately after sleep, which is not entirely unexpected.  The host software closes the JS110 device.  It then opens it again, only to have communication fail again, which is unexpected.  However, the crash is due to an access violation on pubsub unsubscribe, possibly for a callback to one the removed devices that has already been freed.  

We'll try out some other laptops to see if we can duplicate this.