jetperch/pyjoulescope_ui #198: Waveform switches from displaying Voltage data to Current data when zooming in

FirmwareGuy40, 2023-06-29:
### Joulescope model

JS220

### UI version

1.0.22

### What OS are you seeing the problem on?

Windows 10

### What happened?

In the Scope mode display Current and Voltage waveforms at the same time. Both waveforms and statistics look fine at this point.
When zooming in (in one example 26 times or so) the Voltage waveform completely changes from displaying Voltage data to displaying Current data. The statistics units change from Volts to mV.

### What was expected?

When zooming in, the Voltage waveform should continue to display the Voltage data.

### How to reproduce?

1. Open filename.jls  (my files are large, like 900K to 2MB)
2. Scope Mode, enabled Signals I and V 
3. Use the mouse wheel to zoom in around 26 times

### Extra information

Data is captured at: 100 kHz, 2Hz, 1.8uA, 2V

mliberty1, 2023-06-29:
Hi @FirmwareGuy40 - Unfortunately, the GIF did not make it as an email attachment. Could you visit GitHub and edit your response to include the GIF?  Thank you!

mliberty1, 2023-06-29:
Well, that's not right.  The first step for me is ensuring that I am looking at the right things.  I see the date in the filename.  Did you record this with UI 0.10.13?

FirmwareGuy40, 2023-06-29:
Matt,
Yes, UI 0.10.13

Here is the first few lines from the Log file:
*****************************
* Joulescope User Interface *
* UI Version = 0.10.13      *
* Driver Version = 1.0.15   *
*****************************

info = {
  "joulescope": {
    "ui_version": "0.10.13",
    "driver_version": "1.0.15"
  },
  "platform": {
    "name": "win32",
    "python_version": "3.10.9 (tags/v3.10.9:1dd9be6, Dec  6 2022, 20:01:21) [MSC v.1934 64 bit (AMD64)]",
    "platform": "Windows-10-10.0.19045-SP0",
    "processor": "Intel64 Family 6 Model 69 Stepping 1, GenuineIntel",
    "executable": "C:\\Program Files\\Joulescope\\joulescope.exe",
    "frozen": "C:\\Program Files\\Joulescope",
    "paths": {
      "dirs": {
        "app_path": "C:\\Users\\MELLC_Elite2\\AppData\\Local\\joulescope",
        "config": "C:\\Users\\MELLC_Elite2\\AppData\\Local\\joulescope\\config",
        "log": "C:\\Users\\MELLC_Elite2\\AppData\\Local\\joulescope\\log",
        "firmware": "C:\\Users\\MELLC_Elite2\\AppData\\Local\\joulescope\\firmware",
        "themes": "C:\\Users\\MELLC_Elite2\\AppData\\Local\\joulescope\\themes",
        "data": "C:\\Users\\MELLC_Elite2\\Documents\\joulescope",
        "update": "C:\\Users\\MELLC_Elite2\\AppData\\Local\\joulescope\\update"
      },
      "files": {
        "config": "C:\\Users\\MELLC_Elite2\\AppData\\Local\\joulescope\\config\\joulescope_config.json"
      }
    }
  }
}

=====
INFO:2023-06-29 13:07:57,471:logging_util.py:173:root:logging configuration: stream_level=50, file_level=0
WARNING:2023-06-29 13:07:57,351:preferences.py:364:joulescope_ui.preferences:Preference['Oscilloscope']['Device/setting/v_range'] '2V' not valid => '15V'
INFO:2023-06-29 13:07:57,471:main.py:1907:joulescope_ui.main:Arguments: (['C:\\Program Files\\Joulescope\\joulescope.exe', 'ui'],)
INFO:2023-06-29 13:07:57,471:main.py:1908:joulescope_ui.main:Start Qt
INFO:2023-06-29 13:07:57,471:main.py:1910:joulescope_ui.main:Configure high DPI scaling
DEBUG:2023-06-29 13:07:57,781:main.py:1856:joulescope_ui.main:Loaded fonts:
    :/fonts/Lato/Lato-Bold.ttf => 0
    :/fonts/Lato/Lato-Regular.ttf => 1
    :/fonts/Lato/Lato-Black.ttf => 2
INFO:2023-06-29 13:07:57,811:manager.py:133:joulescope_ui.themes.manager:generate images for unchecked
INFO:2023-06-29 13:07:57,841:manager.py:133:joulescope_ui.themes.manager:generate images for checked
INFO:2023-06-29 13:07:57,881:manager.py:133:joulescope_ui.themes.manager:generate images for clickable
INFO:2023-06-29 13:07:58,171:manager.py:133:joulescope_ui.themes.manager:generate images for simple
INFO:2023-06-29 13:07:58,431:manager.py:133:joulescope_ui.themes.manager:generate images for copy
INFO:2023-06-29 13:07:58,551:main.py:1935:joulescope_ui.main:theme load took 0.7800 seconds
INFO:2023-06-29 13:07:59,201:command_processor.py:573:joulescope_ui.command_processor:register command !Device/open
INFO:2023-06-29 13:07:59,201:command_processor.py:573:joulescope_ui.command_processor:register command !Device/close
INFO:2023-06-29 13:07:59,201:command_processor.py:573:joulescope_ui.command_processor:register command !RangeTool/run
INFO:2023-06-29 13:07:59,201:command_processor.py:573:joulescope_ui.command_processor:register command !Widgets/add
INFO:2023-06-29 13:07:59,201:command_processor.py:573:joulescope_ui.command_processor:register command !Widgets/remove
INFO:2023-06-29 13:07:59,201:command_processor.py:573:joulescope_ui.command_processor:register command !Accumulators/reset
INFO:2023-06-29 13:07:59,201:command_processor.py:573:joulescope_ui.command_processor:register command !General/mru_add
INFO:2023-06-29 13:07:59,201:main.py:1034:joulescope_ui.main:_device_change add disable
INFO:2023-06-29 13:07:59,201:usb_inrush.py:98:joulescope_ui.plugins.usb_inrush:USBET20 tool not found - skip usb_inrush plugin
DEBUG:2023-06-29 13:07:59,201:main.py:963:joulescope_ui.main:_device_close: start

mliberty1, 2023-06-29:
Ok, it's JLS v1 recorded, and [here is the issue](https://github.com/jetperch/pyjoulescope_ui/blob/612fa361bf0dffc02dc674be5a83d1e5ec0099d6/joulescope_ui/jls_v1.py#L45).  That copy-paste error populates voltage with the current data.  This also explains why dual-marker statistics would be wrong.  Joulescope UI 1.0 now always computes the right-hand side statistics directly from the displayed data so they should be correct, well, at least matching the displayed waveform.

Thanks for reporting this, and we will be releasing 1.0.23 soon!

mliberty1, 2023-06-29:
We just released [1.0.23](https://download.joulescope.com/joulescope_install/index.html) as beta.  You can download and install it manually.  

You can also have the UI update automatically by switching to the **beta** software update channel:

[image]

Exit the UI and open it, and it will prompt you to update.

FirmwareGuy40, 2023-06-29:
Works wonderfully!!
Nathan

[image]<https://user-images.githubusercontent.com/5393422/249895131-69cd6b2e-7531-401d-a0f1-e63d66c12546.png>