jetperch/pyjoulescope_ui #199: TypeError: 'staticmethod' object is not callable

wangxy0918, 2023-06-30:
### Joulescope model

JS110

### UI version

other

### What OS are you seeing the problem on?

Windows 10

### What happened?

Unexpected Error
The Joulescope UI encountered an error,
and it cannot start correctly.
Please report this error by contacting us: 
[Contact form](https://www.joulescope.com/contact) 
[Joulescope forum](https://forum.joulescope.com/) 
[GitHub](https://github.com/jetperch/pyjoulescope_ui/issues)
Please include the text below,
which has been automatically copied to your clipboard.
--------------------
Exception on Joulescope UI startup. 

Traceback (most recent call last):
  File "D:\code\pyjoulescope_ui\joulescope_ui\main.py", line 652, in run
    ui = MainWindow(filename=filename, is_config_load=is_config_load)
  File "D:\code\pyjoulescope_ui\joulescope_ui\main.py", line 251, in __init__
    self._pubsub.publish('registry/view/actions/!ui_connect', {
  File "D:\code\pyjoulescope_ui\joulescope_ui\pubsub.py", line 606, in publish
    return self._send(cmd, defer=defer)
  File "D:\code\pyjoulescope_ui\joulescope_ui\pubsub.py", line 518, in _send
    return self._process(cmd)
  File "D:\code\pyjoulescope_ui\joulescope_ui\pubsub.py", line 937, in _process
    self._process_inner(cmd)
  File "D:\code\pyjoulescope_ui\joulescope_ui\pubsub.py", line 908, in _process_inner
    return_value = cmds_update_fn[0](self, topic, value)
  File "D:\code\pyjoulescope_ui\joulescope_ui\pubsub.py", line 185, in __call__
    return self.fn(value)
TypeError: 'staticmethod' object is not callable

info = {
  "joulescope": {
    "ui_version": "1.0.23"
  },
  "platform": {
    "name": "win32",
    "python_version": "3.9.5 (tags/v3.9.5:0a7dcbd, May  3 2021, 17:27:52) [MSC v.1928 64 bit (AMD64)]",
    "platform": "Windows-10-10.0.19045-SP0",
    "processor": "Intel64 Family 6 Model 166 Stepping 0, GenuineIntel",
    "executable": "C:\\Python39\\python.exe",
    "frozen": false
  }
}
--------------------

### What was expected?

run

### How to reproduce?

run python -m joulescope_ui

### Extra information

PS D:\code\pyjoulescope_ui> python -m joulescope_ui
Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.23       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.23"
  },
  "platform": {
    "name": "win32",
    "python_version": "3.9.5 (tags/v3.9.5:0a7dcbd, May  3 2021, 17:27:52) [MSC v.1928 64 bit (AMD64)]",
    "platform": "Windows-10-10.0.19045-SP0",
    "processor": "Intel64 Family 6 Model 166 Stepping 0, GenuineIntel",
    "executable": "C:\\Python39\\python.exe",
    "frozen": false
  }
}

=====
WARNING:2023-06-30 16:44:54,605:pubsub.py:1341:joulescope_ui.pubsub:Could not unregister <class 'joulescope_ui.widgets.debug.debug_widget.DebugWidget'> - invalid spec

mliberty1, 2023-06-30:
I successfully duplicated this issue with [Python 3.9.13 - May 17, 2022](https://www.python.org/downloads/release/python-3913/) (the most recent 3.9 with a Windows installer).

It looks like static methods did not become callable until 3.10: [issue](https://bugs.python.org/issue43682)

Because of this, the existing joulescope_ui package 1.0.23 is not compatible with Python 3.9.  I recommend installing and running [Python 3.11.4 - June 6, 2023](https://www.python.org/downloads/release/python-3114/), which we test on.  If you install to "c:\bin\Python3_11_4", here is how:

```
C:\bin\Python3_11_4\python.exe -m pip install -U --upgrade-strategy eager joulescope_ui
C:\bin\Python3_11_4\python.exe -m joulescope_ui
```

Does this work for you?

wangxy0918, 2023-07-02:
> I successfully duplicated this issue with [Python 3.9.13 - May 17, 2022](https://www.python.org/downloads/release/python-3913/) (the most recent 3.9 with a Windows installer).
>
> It looks like static [...]

Can run now. Is it expected with below WARNING?
======================================================================
"
Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.23       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.23"
  },
  "platform": {
    "name": "win32",
    "python_version": "3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)]",
    "platform": "Windows-10-10.0.19045-SP0",
    "processor": "Intel64 Family 6 Model 166 Stepping 0, GenuineIntel",
    "executable": "C:\\Python311\\python.exe",
    "frozen": false
  }
}

=====
WARNING:2023-07-02 23:17:11,858:pubsub.py:1341:joulescope_ui.pubsub:Could not unregister <class 'joulescope_ui.widgets.debug.debug_widget.DebugWidget'> - invalid spec

mliberty1, 2023-07-03:
Yes, this warning is expected.  The warning occurs because the software turns debug mode off when it is already off.  We will add some code to prevent this warning in future versions.
 
I will close this issue, but please reopen if you run into this problem again!