jetperch/pyjoulescope_ui #207: RuntimeError: Could not load file: CHANGELOG.md

plat-zhaw, 2023-07-12:
### Joulescope model

JS220

### UI version

other

### What OS are you seeing the problem on?

Linux Ubuntu amd64 (PC)

### What happened?

Today I updated the Python packages *joulescope-ui* to the newest version 1.0.23 (which is not selectable in the bug report field *UI Version*). I also upgraded the Python package *joulescope* to its newest version 1.1.8. The operating system is Ubuntu 22.04.02 LTS AMD64. When I try and start the Joulescope software,  I get the following error message.

[image]

It appears that some file *README.md* is missing. It follows the terminal output (because I started the software from the command line).

```
plat@6674:~/Documents/mt_smartcard$ pip list | grep -i joulescope
joulescope              1.1.8
joulescope-ui           1.0.23
pyjoulescope-driver     1.3.17
plat@6674:~/Documents/mt_smartcard$ joulescope ui &
[1] 12626
plat@6674:~/Documents/mt_smartcard$ Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.23       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.23"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-46-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/usr/bin/python3",
    "frozen": false
  }
}

=====
qt.dbus.integration: Could not connect "org.freedesktop.IBus" to globalEngineChanged(QString)
WARNING:2023-07-12 11:07:24,285:pubsub.py:1341:joulescope_ui.pubsub:Could not unregister <class 'joulescope_ui.widgets.debug.debug_widget.DebugWidget'> - invalid spec
WARNING:2023-07-12 11:07:24,318:waveform_widget.py:551:joulescope_ui.widgets.waveform.waveform_widget:No default source available
WARNING:2023-07-12 11:07:24,880:pubsub.py:824:joulescope_ui.pubsub:Subscribe to unknown topic registry/JsdrvStreamBuffer:001/settings/duration
Traceback (most recent call last):
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/main.py", line 652, in run
    ui = MainWindow(filename=filename, is_config_load=is_config_load)
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/main.py", line 378, in __init__
    self._pubsub.publish('registry/help_html/actions/!show', 'changelog')
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/pubsub.py", line 606, in publish
    return self._send(cmd, defer=defer)
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/pubsub.py", line 518, in _send
    return self._process(cmd)
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/pubsub.py", line 937, in _process
    self._process_inner(cmd)
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/pubsub.py", line 908, in _process_inner
    return_value = cmds_update_fn[0](self, topic, value)
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/pubsub.py", line 189, in __call__
    return self.fn(pubsub, topic, value)
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/help_ui.py", line 154, in on_cls_action_show
    HelpHtmlMessageBox(pubsub, value)
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/help_ui.py", line 112, in __init__
    title, html = load_help(name, style)
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/help_ui.py", line 73, in load_help
    html = _load_filename(filename)
  File "/home/plat/.local/lib/python3.10/site-packages/joulescope_ui/help_ui.py", line 65, in _load_filename
    raise RuntimeError(f'Could not load file: {filename}')
RuntimeError: Could not load file: CHANGELOG.md
```
I then closed the error message. The graphical user interface showed an empty *Waveform* widget and I could not connect to the Joulescope device. So I closed the software. Unfortunately, I did not take a screenshot, because I thought I could reproduce the error. But the second time I started the *joulescope-ui* it worked just fine, as the following terminal session confirms.

```
plat@6674:~/Documents/mt_smartcard$ joulescope ui &
[1] 13236
plat@6674:~/Documents/mt_smartcard$ Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.23       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.23"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-46-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/usr/bin/python3",
    "frozen": false
  }
}

=====
qt.dbus.integration: Could not connect "org.freedesktop.IBus" to globalEngineChanged(QString)
WARNING:2023-07-12 11:27:18,366:pubsub.py:1341:joulescope_ui.pubsub:Could not unregister <class 'joulescope_ui.widgets.debug.debug_widget.DebugWidget'> - invalid spec
WARNING:2023-07-12 11:27:18,389:waveform_widget.py:551:joulescope_ui.widgets.waveform.waveform_widget:No default source available
WARNING:2023-07-12 11:27:18,767:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-000185/sources/1/info
WARNING:2023-07-12 11:27:32,780:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JsdrvStreamBuffer:001/settings/signals/JS220-000185.v/range
WARNING:2023-07-12 11:27:32,780:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JsdrvStreamBuffer:001/settings/signals/JS220-000185.p/range

[1]+  Done                    joulescope ui
```

I guess the *README.md* is only displayed at the first start after installation/upgrading. So it is nor really a big issue for me. I just thought you might want to know that.

### What was expected?

No error should be shown at start-up of the software.

### How to reproduce?

1. Install or upgrade the Python package *joulescope-ui*.
2. Start the software from the terminal using `joulescope ui`.
3. Marvel at the error message.

mliberty1, 2023-07-12:
Thank you for reporting this issue @plat-zhaw !  I was able to successfully duplicate it.  The CHANGELOG.md file is being included in the [.tar.gz](https://files.pythonhosted.org/packages/da/66/3d399d7a444a136a831d8b9764712c3bdf9e3236b905185bf4aa3b2986fa/joulescope_ui-1.0.23.tar.gz), but it's at the top level.  It needs to also be in the `joulescope_ui` directory.  

For the same reason, **Help** → **Credits** also does not work since it cannot find `CREDITS.html`.

plat-zhaw, 2023-07-12:
You are right about the credits in the help @mliberty1. I just downloaded the application manually from the [website](https://download.joulescope.com/joulescope_install/1/0/23/joulescope_1_0_23.tar.gz). This one does not have the issue. Not at the startup and also not while displaying the credits. See terminal session below.

```
plat@6674:~$ ls -al | grep -i joulescope
drwxrwxr-x  5 plat plat   4096 Jul  4 10:37 .joulescope
drwxr-xr-x 16 plat plat  12288 Jun 29 21:13 joulescope_1_0_23
drwxrwxr-x  3 plat plat   4096 Jul  4 10:37 .joulescope_driver
plat@6674:~$ rm -rf .joulescope .joulescope_driver
plat@6674:~$ ls -al | grep -i joulescope
drwxr-xr-x 16 plat plat  12288 Jun 29 21:13 joulescope_1_0_23
plat@6674:~$ cd joulescope_1_0_23/
plat@6674:~/joulescope_1_0_23$ ./joulescope_launcher &
[1] 18190
plat@6674:~/joulescope_1_0_23$ Publish to unknown topic registry/style/settings/enable
*****************************
* Joulescope User Interface *
* UI Version = 1.0.23       *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.0.23"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.11.4 (main, Jun  7 2023, 11:01:02) [GCC 11.3.0]",
    "platform": "Linux-5.19.0-46-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "executable": "/home/plat/joulescope_1_0_23/joulescope_launcher",
    "frozen": "/home/plat/joulescope_1_0_23"
  }
}

=====
qt.dbus.integration: Could not connect "org.freedesktop.IBus" to globalEngineChanged(QString)
WARNING:2023-07-12 14:52:19,153:pubsub.py:1341:joulescope_ui.pubsub:Could not unregister <class 'joulescope_ui.widgets.debug.debug_widget.DebugWidget'> - invalid spec
WARNING:2023-07-12 14:52:19,280:waveform_widget.py:551:joulescope_ui.widgets.waveform.waveform_widget:No default source available
WARNING:2023-07-12 14:52:19,584:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-000185/sources/1/info
WARNING:2023-07-12 14:52:25,704:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JsdrvStreamBuffer:001/settings/signals/JS220-000185.v/range
WARNING:2023-07-12 14:52:25,704:pubsub.py:887:joulescope_ui.pubsub:Publish to unknown topic registry/JsdrvStreamBuffer:001/settings/signals/JS220-000185.p/range

[1]+  Done                    ./joulescope_launcher
```

I am not sure if that is important or if it helps you at all. I just stumbled across this and thought I should inform you about that.

closed 2023-07-25 by mliberty1