jetperch/pyjoulescope_ui #211: Waveform sometimes vanishes when zooming in

FerriteGiant, 2023-07-21:
### Joulescope model

JS110

### UI version

1.0.23

### What OS are you seeing the problem on?

Windows 10

### What happened?

Waveform and grid vanish while zooming in on the x-axis.

### What was expected?

Waveform and grid should not vanish.

### How to reproduce?

1. Started streaming data, a few seconds later stop streaming data. 
2. Attempt to zoom into waveform in time 
3. Sometimes observe that after some amount of zooming, the waveform and the grid vanish and just the black background is left. 

Waveform and grid becomes visible again when I zoom back out. This only happens sometimes but the only work around I've found is to trash the data and start a new stream.

mliberty1, 2023-07-21:
Hi @FerriteGiant, and thanks for reporting this issue.  I just tried to duplicate it with a JS220 on Win 11 x64, but I could not.  I am traveling now, and I will try with a JS110 when I return next week.  

Upon [inspecting the code](https://github.com/jetperch/pyjoulescope_ui/blob/4d9bed59b9e0c37163de0ddae7703cf2730fdb21/joulescope_ui/widgets/waveform/waveform_widget.py#L3582C9-L3582C9), it (somewhat arbitrarily) limits the maximum zoom Δt range to 1 microsecond.

I have a few questions / requests:
1. How are you zooming in and out?  Are you using the mouse scroll wheel?  The buttons in the waveform control widget?  Something else?
2. Does this happen when you zoom in so far that only 2 samples or less are visible on the screen?  Here is what I see at max zoom with the JS220:
[image]
3. Do you have downsampling enabled?  The default JS110 sample rate is 2 Msps.  When I configured the JS220 for 10 kHz, I was able to get the signal traces to disappear, but the grid lines were still present.
4. When you continue to zoom and out after this occurs, does the Δt value at the right of the x-axis line continue to update?
5. If this is relatively easy for you to duplicate, I would love to get an error report using our new Joulescope UI alpha version.  [Download and install 1.0.25](https://download.joulescope.com/joulescope_install/index.html).  Duplicate this issue.  Then select  **Help** → **Report Issue**.  Please mention this issue in the description.

FerriteGiant, 2023-07-21:
> 1. How are you zooming in and out?  Are you using the mouse scroll wheel?  The buttons in the waveform control widget?  Something else?

Mouse scroll wheel only

> 2. Does this happen when you zoom in so far that only 2 samples or less are visible on the screen?  Here is what I see at max zoom with the JS220:

I noticed that as well but as a separate thing. 

> 3. Do you have downsampling enabled?  The default JS110 sample rate is 2 Msps.  When I configured the JS220 for 10 kHz, I was able to get the signal traces to disappear, but the grid lines were still present.

When I was noticing this issue I was set to 2 Msps

> 4. When you continue to zoom and out after this occurs, does the Δt value at the right of the x-axis line continue to update?

Not sure, I'll keep a look out for that the next time I see this happen.

> 5. If this is relatively easy for you to duplicate, I would love to get an error report using our new Joulescope UI alpha version.  [Download and install 1.0.25](https://download.joulescope.com/joulescope_install/index.html).  Duplicate this issue.  Then select  **Help** → **Report Issue**.  Please mention this issue number 211 in the description.

I tried a few times to repro it today and I wasn't able to get it to happen within half-a-dozen attempts. Next time it happens I'll grab a screen recording. 

Note: One of my colleagues also had issues with this on Windows 10 with the JS220.

mliberty1, 2023-07-24:
I think that I finally successfully duplicated this issue:

[image]

The log shows this issue:
```
ERROR:2023-07-24 13:57:30,481:waveform_widget.py:1284:joulescope_ui.widgets.waveform.waveform_widget:Exception during drawing
Traceback (most recent call last):
  File "C:\repos\Jetperch\pyjoulescope_ui\joulescope_ui\widgets\waveform\waveform_widget.py", line 1282, in plot_paint
    self._plot_paint(p, size)
  File "C:\repos\Jetperch\pyjoulescope_ui\joulescope_ui\widgets\waveform\waveform_widget.py", line 1356, in _plot_paint
    if not self._draw_x_axis(p):
           ^^^^^^^^^^^^^^^^^^^^
  File "C:\repos\Jetperch\pyjoulescope_ui\joulescope_ui\widgets\waveform\waveform_widget.py", line 1485, in _draw_x_axis
    x_grid = axis_ticks.x_ticks(x_range64[0], x_range64[1], major_count_max)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\repos\Jetperch\pyjoulescope_ui\joulescope_ui\widgets\waveform\axis_ticks.py", line 184, in x_ticks
    'offset_str': time64.as_datetime(k).isoformat(),
                  ^^^^^^^^^^^^^^^^^^^^^
  File "C:\bin\Python3_11_3\Lib\site-packages\pyjoulescope_driver\time64.py", line 98, in as_datetime
    return datetime.datetime.fromtimestamp(t, tz=datetime.timezone.utc)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument
```

Clicking the **Zoom all** button seems to have fixed the behavior so that I can zoom in again.  Hopefully, this will be a consistent workaround.  

I still have no idea what I did to cause this behavior.  I will instrument the code and try to duplicate this to get more information about what is happening.

FerriteGiant, 2023-07-24:
yep, that looks like it!

mliberty1, 2023-07-25:
Fixed in 1.0.27

closed 2023-07-25 by mliberty1