jetperch/pyjoulescope_ui #213: Dual marker shows wrong values

atsju, 2023-07-21:
### Joulescope model

JS110

### UI version

other

### What OS are you seeing the problem on?

Windows 10

### What happened?

UI v1.0.25 

I observed a discrepancy between dual marker values and window value. They should be almost the same because signal is 0 outside markers.
[image]

I exported both "visible data" and "all data" and when I opened the files, the value for markers has changed. For me this means there is a bug in dual marker integral.
[exportedJLSfiles.zip](https://github.com/jetperch/pyjoulescope_ui/files/12125615/exportedJLSfiles.zip)

### What was expected?

Correct computed values in dual marker

### How to reproduce?

capture some data and put dual marker. I did not do anything special and I'm not sure this is reproductible. I will update the issue if I find something.

mliberty1, 2023-07-21:
Hi @atsju - I don't know about this particular waveform.  It looks like there could be a small negative current event at about 925, but you can double-check.  On the marker with text, right-click, then select **Show statistics** → **Left**.  Drag the right dual marker all the way to the right side and the left dual marker all the way to the left side.  Do the values now match?  If so, then this is not an issue with the UI.  Instead, it is a feature of the waveform.

If you do not want to adjust your measurement dual markers, you can place additional dual markers to the left and right that go to the display edge.  You can sum up the integrals.

What do you find?

atsju, 2023-07-21:
Please open the zip file. I shared the exact same file. And you should see that value is different from the print screen.

In the mean time I will do what you request. Lucky you, I didn't close the capture

atsju, 2023-07-21:
UI bug confirmed :  
[image]

Please note that right click on a thin maker line is not easy and not natural. I did not expect this feature there.

mliberty1, 2023-07-21:
Did you recreate this on the buffered data or the one of the JLS files that you provided?

atsju, 2023-07-21:
this capture is from the same instance as first capture with same data and window not moved. Opening an export of same data does not show the bug (as you can see in the zip)

mliberty1, 2023-07-21:
So far, I have not been able to recreate what you are seeing.  However, I am also using a JS220, not a JS110.  I do expect small differences between the dual markers and the right-hand side statistics, like this:
[image]

This does *not* explain the 10% difference in the second image above.

It sounds like this is easy data to capture.  Since I only have access to the JLS file (which does not show this issue), can you check if this is a consistent issue?  Capture data, then place dual markers over one event, as you did above.  Repeat for a couple other events.  Is the value dual marker value and the right-hand side statitics values always the same as you captured above (dual marker integral approximately +10% greater than right-hand side statistics)?   

Note the these statistics values are computed differently.  The dual markers are computed over the exact marker positions, which is the pixel center when repositioning the markers.  In contrast, the right-hand side statistics are computed over the visible plot data, which is left-most sample in the first pixel to the right-most sample in the last pixel.  To make things more complicated, the buffered sample data and JLS data compute their statistics using completely different code.  

For performance reasons, this code is highly optimized which makes it complicated.  For example, [here is the code](https://github.com/jetperch/joulescope_driver/blob/574d309ea672cc12d95df0c42b1678daae9f893f/src/buffer_signal.c#L604) for the buffered sample data.  The code for the JS110 and JS220 is the same, but how it is used could be slightly different.  

I am traveling now, and I will try some more testing with a JS110 when I return to the office next week.

atsju, 2023-07-24:
For information, I did some more tests right now with same device and it's definitely reproductible.
[image]

mliberty1, 2023-07-24:
I successfully duplicated this issue:

[image]

I was able to track it down to the summary computation in [joulescope_driver](https://github.com/jetperch/joulescope_driver).  These summaries are used to greatly speed the computation of statistics and waveforms.  Unfortunately, when more than one summary value was computed over a single input block, the statistics accumulator was not reset correctly.  See [joulescope_driver 8c6714d](https://github.com/jetperch/joulescope_driver/commit/8c6714de856e8ecf6802a537063f37bf32c963ee#diff-721f8b5508a475e2d894f8a0c82dd34720277d42f32b1b697f9e45cf5e0f445f).  This error also was likely slightly distorting zoomed out waveforms for live streaming and buffered samples.  It would not have affected JLS files.

This will be fixed in joulescope_driver 1.3.18 and Joulescope UI 1.0.26.

mliberty1, 2023-07-25:
You should be able to download and install [1.0.26](https://download.joulescope.com/joulescope_install/index.html), which fixes [this issue and more](https://github.com/jetperch/pyjoulescope_ui/blob/main/CHANGELOG.md).  The new relative x-axis markers do not quite work correctly, so we will make a new release 1.0.27 that will hopefully make it to our next stable release.

closed 2023-07-25 by mliberty1