jetperch/pyjoulescope_ui #218: button proposal (cosmetic change)

atsju, 2023-08-04:
### UI version

1.0.28

### Your idea

I'm using new UI for some time now and it's still a bit painfull for me because it's less intuitive than 0.x.y (to me).

Here are 2 buttons I would like to change.
First one because you never know if it's powered or not (red VS green).
Second one because I did not find it when I needed it.
[image]

For power button, I think something in line with this would be clearer:
[image]

For second button, a very simplified version of this would probably be very clear:
[image]
[image]

mliberty1, 2023-08-04:
Thank you for the feedback @atsju .  I appreciate the feedback on making the power on/off more than just a color change.  For possible future compatibility, I like power on/off rather than current flow on/off (like open/close switch).  What about a backslash through the icon, like this:
[image]

I am more conflicted about the other one, which is why it is currently just green / black.  This button opens and closes the instrument.  It does not unplug it, reset it, or completely disable it.  When open, this Joulescope UI instance opens the instrument for exclusive access.  When closed, the Joulescope UI closes the instrument, and that instrument can be opened by another program.  Power connection/disconnect is not the right icon for this.  Open/close often references doors, locks, chests, and folders.  Unfortunately, I don't like any of these concepts for this.

However, I took another look at The Noun Project, and I like this concept for open:
https://thenounproject.com/icon/open-3560617/

I could blend that with our existing device icon (minus the dots), perhaps for something like this:
[image]

What do you think?

atsju, 2023-08-04:
> What do you think?

I don't find the device_open/close very explicit either. 
I understand the open/close feature is not plug/unplug but in the end you can only "connect" one UI to une joulescope. So it may not be technically accurate but I'm not sure it's a problem.

The crossed power is OK'ish. At least better than existing.

mliberty1, 2023-08-04:
The notion of plug / unplug has many possible interpretations for Joulescopes and measurement systems.  I already have a good idea of the type and quantity of support I would have to deal with if we added this icon for device open/close:
[image]

I would much rather leave it green / black and have people read the tooltip description since 90% of Joulescope customers should never even use this feature.  It will become even less important when the UI includes full simultaneous, multiple Joulescope support in the fall.

Any other proposed device open/close icons?

atsju, 2023-08-04:
> Any other proposed device open/close icons?

Nope. thank you.

atsju, 2023-08-04:
Also, we do not understand why all the buttons are blinking. This one especially does not need to blink.

mliberty1, 2023-09-19:
Addressed in 1.0.30