jetperch/pyjoulescope_ui #219: Crash after Waveform save

GilWu, 2023-08-26:
### Joulescope model

JS220

### UI version

other

### What OS are you seeing the problem on?

Windows 10

### What happened?

I recorded a Waveform with 50 Hz for ~3 minutes. When saving this recording (complete data and only visible data), the files are created and afterwards the application crashes.
When trying to open the file, the application not even starts and instead suggests to report a bug.

### What was expected?

UI should not crash and saved data should be displayed. 

### How to reproduce?

1. 50 Hz (V,I,P, GPI0, GPI1)
2. Record in Waveform view for ~ 3 Minutes
3. Right click and select "Expert ..."

### Extra information

When trying to save a much shorter recording (just a few seconds), it works.

UI 1.0.29

mliberty1, 2023-08-28:
Hi @GilWu !  I tried to duplicate this issue a few times, but I was unsuccessful.  Here is what I did:

[image]

Does this match what you did?  

I did notice that the GPI signals are not being downsampled, so they are still at the full 2 Msps rate.  This was intentional at one point to maintain time resolution, but we may want to revisit this decision.

Can you report this issue from the UI?  If you haven't started the UI more than 2 times since the issue occurred, you can just open the Joulescope UI and select **Help → Report Issue**.  In the description, please mention GitHub issue #219.  Otherwise, please duplicate the issue, and then report the issue.  Reporting the issue will submit more details that we can investigate.

GilWu, 2023-09-03:
Hello @mliberty1 ,

yes, this matches exactly to what I did. I cannot send the error log at the moment because a long-time recording is running, but I can share with you the file which doesn't like to open. 
[Full_Cycle.zip](https://github.com/jetperch/pyjoulescope_ui/files/12506600/Full_Cycle.zip)

mliberty1, 2024-02-26:
We have been unable to duplicate this issue.  Joulescope UI 1.0.56 contains a bunch of fixes around Qt QObject memory management that were likely the cause of this issue.  We also improved the JLS recovery so that it should be able to open files that were not properly closed.

We are going to close this issue with the 1.0.56 release, but please reopen if you see this again!

closed 2024-02-28 by mliberty1