jetperch/pyjoulescope_ui #222: Waveform widget have option to maintain Δt on sample stream start

mliberty1, 2023-09-15:
### Joulescope model

JS220

### UI version

1.0.28

### Your idea

The Joulescope UI 0.10.x and earlier maintained the Δt window when restarting sample streaming.  It also forced the waveform to be pinned to the right.

With the new Joulescope UI 1.0, the Waveform widget does not maintain the Δt view.  Instead activates both pin-left and pin-right when restarting sample streaming.  While this is great for seeing all captured data, it does alter the time scale of the Waveform widget.  If a user likes the Δt setting for watching events scroll by, they will need to manually configure Δt.

The UI should add an option to prevent pin-left and keep the current time range on sample streaming start.

Doug-Nocturnal, 2024-11-13:
Has there been any progress on adding this option? My team has been using 0.10.x versions since this behavior is very counterintuitive for those of us who are used to an oscilloscope in roll mode.

mliberty1, 2024-11-14:
I looked into this a while back, and this feature would add even more complexity to some already complex code controlling the time view.  I was considering closing this as "will not fix" with the upcoming 1.2.1 release.

It sounds like I do not fully appreciate why this feature is an improvement, and your team finds the existing behavior counterintuitive.  I also looked back at our email chain.  If I remember, this was a minor concern that just operated differently from some other equipment you had.  I suspected you would get used to the Joulescope UI approach soon enough.

It sounds like I don't fully understand the reason that this behavior is so counterintuitive.  Then again, I don't think that I have ever used an oscilloscope in roll mode, either.

Can you explain how you are using the Joulescope UI and how maintaining Δt at sample streaming start would help?

Doug-Nocturnal, 2024-11-19:
Thanks for the reply. We develop very low power devices with no user interfaces and often use the Joulescope as a means to determine the state and operation of a device by only looking at the streaming current waveform. We can easily tell when our devices are sleeping, writing to flash, sampling from the ADC, sending data via Bluetooth, etc but just looking at the current waveform. 
The shape and duration of the waveform current peaks is very important as each device function has its own unique signature. With the variable Δt you get when pinning the data it compresses the data in the X axis as the screen fills up and causes everything to look distorted. For example a sine wave would look like its increasing in frequency as the display accumulates. With the older software and the fixed Δt, a single event of our device would have a consistent width on the screen.

mliberty1, 2024-11-20:
We will take another look at potentially implementing this with the next release.  UI 1.2.1 is almost done with testing, so it will be after that.