jetperch/pyjoulescope_ui #224: [enhancement] automate cursor statistics best position

atsju, 2023-09-28:
### Joulescope model

JS220, JS110

### UI version

1.0.29

### Your idea

By default, the statistics from cursors should choose where they display. In double cursor mode, there are 4 possible positions (left and right of left and right cursors).

In order of priority cursor should not be masked by end of window like in capture, and then it could also check where least signal (yellow) would be masked.
[image]

mliberty1, 2023-09-28:
I can see the benefit of the statistics text automatically selecting the right or left side of the marker to keep the text in the visible area.  This works for both dual markers and single markers.

I am concerned about jumping from marker to marker for dual markers, though.  Anything that "jumps" increases confusion and frustration.

Here is how "auto" staying on a single marker looks for dual markers:
[image]

I think that this is an improvement that should be the new default.  What do you think?

mliberty1, 2023-10-28:
Addressed in 1.0.35.

closed 2023-10-28 by mliberty1