jetperch/pyjoulescope_ui #226: Measurements that disappear when the joulescope is disconnected

atsju, 2023-10-04:
### UI version

1.0.31

### Your idea

Hi Matt,

This has been written by my coworker. We absolutely did not discuss together about this before but it bothered me exactly the same. Just not enough to open an issue on my side.
https://forum.joulescope.com/t/measurements-that-disappear-when-the-joulescope-is-disconnected/659

```
1. How often does this behavior occur that you accidentally unplug your Joulescope?
2. Are you actually unplugging your Joulescope, or are you finding that plugging/unplugging other devices on USB is causing your Joulescope to disconnect? If so, can you provide more detail about your USB topology including the use of hubs, docks, & adapters?
3. What type of environment are you working in where this seems to occur: your desk, while traveling, in a lab?
4. Are you frequently unplugging other devices (like your device under test, debug adapters, etc)? Simply labeling the Joulescope USB connector may significantly help in this case.
```

1. For me about 1 to 5 times a day. It's not about unplugging the Joulescope but rather touching the USB  with my mouse that disconnects the Joulescope for some milliseconds.
2. No related to other devices
3. Not happening on desk where we have space, second screen and additional mouse/keyboard. It happens on smaller desk when we need to be near an oven or other specific instrument on a smaller desk.
4. Unrelated

A popup "connection lost" would be annoying but could be a solution. Then either save data or drop them as you already do. Reconnection could happen only after user made his choice.

atsju, 2023-10-04:
For me it was just bumping on the USB cable (on computer side of JS110) for him it was bumping on JS220 USB-C side.
We just did some additional tests by playing with USB cables on both sides of both JS110 and JS220 and got no disconnection. Very strange.

Let's put this into standby for now. We will try to reproduce and add information if it happens again. If you don't hear from us within 1 month, consider closing the issue.

mliberty1, 2024-02-28:
Hi @atsju - Is this still a problem that we should address?