jetperch/pyjoulescope_ui #227: Range display issue when JS110 and JS220 on same interface

atsju, 2023-10-04:
### UI version

1.0.29

### What OS are you seeing the problem on?

Windows 10

### What happened?

Range display is wrong (offset by 1) when both JS110 and JS220 are displayed at same time on interface

### What was expected?

correct display of range

### How to reproduce?

1. plug a JS110 and a JS220 is any order.
2. have 0 current so that each joulescope is in range 18µA
3. observe the range display

### Extra information

[image]
[image]

mliberty1, 2023-10-04:
So, this issue is very annoying.  The JS110 has a separate **2 A** range, but the JS220 does not.  All other ranges are the same.  Internally, the range is represented as an integer.  So JS110 current range integers are not comparable to JS220 current range integers.  Future Joulescopes may have even different current ranges.

Supporting the current range display becomes more confusing with our upcoming official support for multiple simultaneous Joulescopes.  You are trying that now.  While it is somewhat usable today, the Waveform widget in particular lacks controls and a legend to make it fully usable.

Perhaps the right solution is to convert the current range into a floating point number and display it on a log y-axis scale?  When a single Joulescope is present, we can use the existing labeling.  When incomparable instruments are connected, it can switch to current range values in amps on a log scale.  We will lose the explicit labeling for each current range, but at least it won't be wrong.  What do you think?

atsju, 2023-10-04:
As a developer I think understand the limitations you are facing but as an end user I don't care and just want it to work.

What about having 2 scales in same widget ?

mliberty1, 2023-10-04:
The easier and "correct" solution is to undo #162 and just display the current range integer when you have different device types present in the same Waveform widget.  If you don't like it, you can open two Waveform widgets and assign Joulescopes of each type to each widget.  You can't do this today, but you will once we complete simultaneous Joulescope support.

I am not sure where we would put 2 scales in the same Waveform widget.  The layout not designed for that.

Another solution is to combine the JS110's 2 A and 10 A range into a single 10 A range, which is actually what the JS220 does on-instrument.  Both use the same 0.01 Ω shunt resistor, just with different analog gain.

In thinking about this more, a problem with log scale is displaying "off" in any sensible way.

Is this really a problem?  How often do you actually want to see the current range?

atsju, 2023-10-05:
> How` often do you actually want to see the current range?

Always. Because I need to know the precision of my measurement.

> Is this really a problem?

only when using JS220 and JS110 simultaneously. Please do not undo #162. Merging 2A and 10A range would probably be less intrusive (for me because I don't use these ranges) while waiting for a way to open 2 different waveform widgets. After that is possible a solution that will not make people ask "why do I never see 2A range" would be preferred.

mliberty1, 2023-10-05:
Ok, I think I am now starting to understand.  You do not want to see the current range.  You really want to see the measurement error.  You are only using the current range as an estimator for measurement error.  

The Joulescope JS220 and Joulescope UI really do not help you understand measurement error, which is common practice for test equipment.  You usually get datasheet specs, but then no assistance through the display or UI in estimating measurement error.  Understanding measurement error is therefore often a paper exercise.  While understanding measurement error is necessary for sound engineering, we often "trust" (after verifying) that the instrument is accurate enough.  When the instrument is marginally accurate enough, engineers need to do more work to quantify & understand measurement error.

I have a bunch of questions:

1. What measurement errors are important to you?
    a. Pixel-by-pixel error bars in the Waveform widget plots?  
    b. The error for the average current in Waveform widget dual markers?  
    c. The error for the average current in the Waveform widget right-hand side statistics?  
    d. The error for each Multimeter current reading?
    e. Something else?
2. When you are trying to estimate the current measurement error, is the instrument in a fixed current range over the region of interest?  Alternatively, are you trying to estimate dynamic measurement error across current range changes?
3. How accurate do your measurement error estimates need to be?  Joulescopes are designed to be more accurate than the typical unit-to-unit variations you will see.
4. Have you encountered situations where the Joulescope JS220 was not accurate enough for your measurement?  Why & what additional information can you provide?

Feel free to respond here or by email.  We can also set up a time to talk live if that is easier.

atsju, 2023-10-05:
For the simple explanation we are often measuring sleep current of IoT devices 1-10µA and it's very important to check that the JS has switched back to 18µA range after a peak of device activity. Sometimes it can be a little long. Sometimes it will not happen due to external noise sources. So I always have the "range" open to understand what happens. 

Sometimes we need to know thee range to know the shunt resistor that is connected.

For our dream device accuracy, I will send you an email.

mliberty1, 2023-10-05:
Hi @atsju  - This makes sense.  So it's less about measurement accuracy and more about ensuring that sleep mode is measured in the 18 µA range.  

This makes senses.  Thank you for explaining, and thanks for the email!

mliberty1, 2023-12-07:
As of Joulescope UI 1.0.45 (currently alpha), the current range y-axis labels match the selected device (source):

[image]

closed 2023-12-07 by mliberty1