jetperch/pyjoulescope_ui #228: Auto scale of the range windows

Nono-73, 2023-10-04:
### Joulescope model

JS220

### UI version

1.0.31

### What OS are you seeing the problem on?

Windows 10

### What happened?

**First issue:**
When select Rightclick/ Range / Auto on the range windows, the range appears on the top of windows, not in the middle like current or voltage. 
[image]

**Second issue:**
In manual scale, When I zoom out using the mouse cursor, the scale disappears. This happens on all windows (Current, Voltage, Power, Range). 
[image]

### What was expected?

For the second issue I recommend a maximum scale. The joulescope cannot measure kA...
[image]

### How to reproduce?

1. ...
2. ...
3. ...

mliberty1, 2023-10-04:
Hi @Nono-73

# Issue 1

This issue occurs because there is no variation on the current range waveform.  The existing code was zooming to a 1e-9 range in this case.  However, it performed math on 32-bit floating points, which was not enough to accurately represent the zoom range 1e-9 with the average value 5.   32-bit float only has a 23-bit mantissa, which represents 1.2e-7 relative minimum increment.  I changed this to fix the maximum current range zoom and also I changed this to compute the a zoom that is sufficiently represented within 32-bit floating point math.  Fixed.

# Issue 2

The labels disappearing on the current waveform is due to the waveform widget attempting to display grid lines outside of the valid range.  Here is the error:

```
  File "C:\repos\Jetperch\pyjoulescope_ui\joulescope_ui\widgets\waveform\waveform_widget.py", line 1635, in <listcomp>
    y_grid['labels'] = [_JS220_AXIS_R[int(s_label)] for s_label in y_grid['labels']]
                        ~~~~~~~~~~~~~^^^^^^^^^^^^^^
```

We can fix this to not throw an error.   However, zooming to display more than the valid set of current range values is really not useful.

I do not see the labels on the other plots disappearing when zoom out.  Instead, the labels eventually all become 0.  This was due to a computational error for how the code was forcing zero to display as zero and not some floating point residuals.  Fixed.

# Issue 2b

I believe that the request here is for each plot to implement a meaningful upper & lower bound.  Ideally, these would come from the connected sources that are being displayed.  However, we can pick something since no Joulescope product has plans to support TA (10^12).

We do have plans to allow for scaling factors for voltage dividers and external current shunts.  To allow for these, we need ranges bigger than the JS110 and JS220 support.  I've added ±50 A, ±250 V, ±1000 W.  You will need to select **File** → **Clear config and exit** to get these when the next Joulescope UI release.  Fixed.

All of these fixes were in [79e27be](https://github.com/jetperch/pyjoulescope_ui/commit/79e27bebc9d41fa1209582c932c28e0d4fac665b) and will be included in the next official release.

mliberty1, 2023-10-28:
Addressed in 1.0.35.

closed 2023-10-28 by mliberty1