jetperch/pyjoulescope_ui #229: Segmentation fault on Ubuntu when starting Oscilloscope view

franslundberg, 2023-10-04:
### Joulescope model

JS220

### UI version

1.0.31

### What OS are you seeing the problem on?

Linux Ubuntu amd64 (PC)

### What happened?

Multimeter view works, but when choosing Oscilloscope View, the application crashes with a segmentation fault. Output from terminal follows below.

    f@f30:~/Downloads/joulescope_1_0_29$ ./joulescope_launcher 
    Publish to unknown topic registry/style/settings/enable
    *****************************
    * Joulescope User Interface *
    * UI Version = 1.0.29       *
    *****************************
    
    info = {
      "joulescope": {
        "ui_version": "1.0.29"
      },
      "platform": {
        "name": "linux",
        "python_version": "3.11.4 (main, Jun  7 2023, 11:01:02) [GCC 11.3.0]",
        "platform": "Linux-6.2.0-33-generic-x86_64-with-glibc2.35",
        "processor": "x86_64",
        "executable": "/home/f/Downloads/joulescope_1_0_29/joulescope_launcher",
        "frozen": "/home/f/Downloads/joulescope_1_0_29"
      }
    }
    
    =====
    WARNING:2023-10-04 19:17:42,058:pubsub.py:899:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-001502/sources/1/info
    Segmentation fault (core dumped)

### What was expected?

Application should not crash.

### How to reproduce?

1. Run on Ubuntu.
2. Choose Oscilloscope view.

### Extra information

Tested on 1.0.29 and 1.0.31 with same result. OS: Ubuntu 22.04.3 LTS.

mliberty1, 2023-10-04:
The Joulescope UI definitely runs on Ubuntu 22.04 LTS.  Since the application starts, the basics are good.  The difference with the Oscilloscope view is that it uses OpenGL.  Does this computer have a good graphics setup that supports OpenGL?

Can you try something to further isolate the issue?  Start UI 1.0.31.  If you stay in the Multimeter view and select Widgets → Waveform, do you get the same crash?  If so, can you submit the full crash report from within the UI?

franslundberg, 2023-10-08:
If I stay in Multimeter view and select Widgets - Waveform, it works as expected. No crash.

mliberty1, 2023-10-08:
> If I stay in Multimeter view and select Widgets - Waveform, it works as expected. No crash.

That is surprising to me.  Has switching to Oscilloscope view ever worked.  If this is a new behavior, you can select **File** → **Clear config and exit** to revert to UI the defaults.

If this failure has always been the case, can you duplicate it and submit the full crash report from within the UI?

mliberty1, 2023-10-28:
Hi @franslundberg - we just published [1.0.35](https://www.joulescope.com/pages/downloads).  Can you download this and see if the issue occurs?

If it still occurs, can you submit the issue report from within the UI?  If the issue report does not pop up automatically on this failure, please select Help -> Report Issue the next time you launch the Joulescope UI.  In the description for the issue report, please note GitHub issue #229.  Thank you!

mliberty1, 2023-12-09:
We have been unable to duplicate this behavior.  As far as I know, we have not received a crash / issue report from @franslundberg , so I am going to close this issue.  Please reopen if you continue to observe the problem with Joulescope UI 1.0.46, and please submit an issue report and refer to issue 229.

closed 2023-12-09 by mliberty1