jetperch/pyjoulescope_ui #230: high CPU usage

atsju, 2023-10-06:
### UI version

1.0.29

### Your idea

Tested with JS110 on i7-10850H (6 core / 12 threads)

UI in acquisition : 10-11% CPU usage
UI in pause : 5-7% CPU usage
UI does nothing, no JS connected : 1-3% CPU usage

I don't know how the Joulescope communicates with application exactly but when it's not even connected, 3% CPU usage is unacceptable. It should be near 0.
It would be great if it could also be nicer when in pause.

### Does your idea concern a specific OS?

Windows 10

mliberty1, 2023-10-06:
How did you measure these?  The CPU percentages in Windows **Task Manager** are now based upon the percentage currently available.  With CPU throttling, this number is meaningless.

A better tool is [**Process Explorer**](https://learn.microsoft.com/en-us/sysinternals/downloads/process-explorer) which still displays CPU utilization as percentage of full load.

Here is my computer showing both while in Oscilloscope view with once connected, streaming JS220:
[image]

What do you find with **Process Explorer**?  Are you happy closing this issue?

mliberty1, 2024-02-28:
I looked into this.  We did switch from Pyinstaller to Nuitka back in November 2023, at least for Windows.  Nuitka should help reduce CPU usage by compiling python code into native code (but still using the python C API).  With 1.0.58, SysInternals Process Explorer shows less than 1% average on three different Windows 11 amd64 (x86-64) computers in acquisition mode.

When the UI is paused, it stops storing signals to the sample buffer.  However, all connected Joulescopes keep streaming data.  This allows recordings to continue when paused.  The host computer will still be busy handling all the USB traffic.  We could change the behavior to stop connected Joulescopes from streaming while paused, but it will prevent a whole class of features for very little upside.  

When the UI is idle with no Joulescopes connected, the remaining usage is actually due to the blinking buttons.  The Qt style polish/unpolish takes a small amount of processing.  It's actually a good UI feature to disable these buttons when no devices are connected, so we will make this change for the next release.  The UI will still have the internal blink events, but we ave processing by skipping the UI repaint.

Note that we use WaitForMultipleObjects on Windows, no polling loops.

I am going to close this issue, but please comment or reopen if you think CPU usage is still a problem.

closed 2024-02-28 by mliberty1