jetperch/pyjoulescope_ui #232: Saved data is "truncated" if app is closed before recording stopped.

attie-argentum, 2023-10-23:
### Joulescope model

JS110

### UI version

1.0.31

### What OS are you seeing the problem on?

Windows 10

### What happened?

I was recording a long waveform, and accidentally closed the application.

I was relieved to find that the event had been captured ~20 seconds before my mistake... then quickly upset to find that the `*.jls` file wouldn't open, and the application reported the data was truncated.

### What was expected?

The recording was stopped and the file was closed cleanly.

### How to reproduce?

1. Open JouleScope software
2. Start a recording
3. Quit the software (e.g: using the X in the top-right corner of the window)
4. Attempt to open the recording

### Extra information

[image]

mliberty1, 2023-10-23:
Hi @attie-argentum - sorry to hear that this is not working for you.  I just tried, and I was able to duplicate this behavior.  Starting with Joulescope UI 1.0.30, we use pyjls 0.8 which is supposed to be able to restore truncated files.

I took a very quick look at the code using a JLS file captured this way.  It does not handle file truncation in the case that the file was never closed and errors out before getting to the truncation recovery code.  I tried a very quick fix, but I was unsuccessful.  I appears to repair the file, but it still will not open correctly.  I will take a closer look tomorrow.

closed 2023-10-23 by mliberty1

attie-argentum, 2023-10-23:
No problem - many thanks for the quick response! It's a minor / edge-case bug, but still one that just bit me. 🙃