jetperch/pyjoulescope_ui #235: Joulescope flagged as virus by G DATA

jonathanlarochelle, 2023-10-30:
### Joulescope model

JS110

### UI version

other

### What OS are you seeing the problem on?

Windows 10

### What happened?

Execution of file Joulescope.exe is blocked by GData Antivirus due to detected presence of Gen:Variant.Tedy.474701.

### What was expected?

Execution of Joulescope.exe should not be falsely blocked by an antivirus software.

### How to reproduce?

1. Run G DATA Security Client
2. Try to execute Joulescope.exe
3. Observe that the G DATA Monitor blocks the execution.

### Extra information

According to VirusTotal, 16/72 Antivirus services detect a virus in Joulescope.exe: https://www.virustotal.com/gui/file/c2ff95bfb6e5b9570524651191a3731588da70b7d7634008b452fff5d5fcdd05/details

Current workaround is placing Joulescope.exe in the list of exceptions. This is however not ideal in a corporate setting, where exceptions are undesired and/or fully prohibited, preventing the use of the Joulescope UI.

mliberty1, 2023-10-30:
Please report this file as a false positive to GData.  I have already reported it to Microsoft and BitDefender.  Please see the full post on the [forum](https://forum.joulescope.com/t/joulescope-ui-1-0-36-flagged-by-anti-virus-software/673).

Unfortunately, this is most likely not a problem with the Joulescope UI, but with PyInstaller being flagged as malicious.  We will continue to work with the anti-virus vendors to get this likely false positive cleared.

jonathanlarochelle, 2023-10-30:
Thank you for the fast answer and the detailed forum post. I will go back to 1.0.31 and try again in a couple of days.

mliberty1, 2023-11-17:
Fixed by 1.0.42.  

Joulescope UI 1.0.42 ([download](https://download.joulescope.com/joulescope_install/index.html)) scans [clean on VirusTotal](https://www.virustotal.com/gui/file/35277c60b8c799053982eb5b69b11f5a8cb57cbe363fa2b30466a349dae209f9/detection)!

In addition to fixing the anti-virus false positive detections, this release also includes the new JS220 soft-fuse and offset calibration. You can view the full [changelog](https://github.com/jetperch/pyjoulescope_ui/blob/main/CHANGELOG.md).

We solved this issue by switching our distribution approach from [PyInstaller](https://pyinstaller.org/) to [Nuitka](https://nuitka.net/). We are also now a proud Nuitka sponsor with full access to Nuitka commercial, which will also help us [better manage](https://nuitka.net/doc/user-manual.html#windows-virus-scanners) any false positive anti-virus detections in the future.

closed 2023-11-17 by mliberty1

jonathanlarochelle, 2023-11-20:
I can confirm that there is no issue with GData. Thank you for the quick fix.