jetperch/pyjoulescope_ui #236: Joulescope UI 1.0.31 is not creating an .exe

chriszylstra, 2023-11-02:
### Joulescope model

JS220

### UI version

1.0.31

### What OS are you seeing the problem on?

Windows 11

### What happened?

When I update the UI to the latest stable 1.0.31 I get an installation error shown below:

[image]

The installer is not creating the joulescope.exe in the filepath C:\Program Files\Joulescope\joulescope.exe

[image]

The problem is solved by running the old installer for v1.0.30 and I can use the software

### What was expected?

I was expecting the installer to create the joulescope.exe at the file location C:\Program Files\Joulescope\joulescope.exe
[image]

### How to reproduce?

1. Run the joulescope_step_1_0_31.exe
2. Observe the filestructure C:\Program Files\Joulescope\

### Extra information

I am not uninstalling between tests. Only running the old 1.0.30 and new 1.0.31 Joulsescope UI installation file.

chriszylstra, 2023-11-02:
Just reading through the open/closed issues and releases to see if there are other's having this issue and I notice this:
[image]

so I'll test 1.0.37 beta and see if this comment is related

mliberty1, 2023-11-02:
Hi @chriszylstra - Recent anti-virus software changes flagged joulescope.exe as malicious.  It is likely that it quarantined this file. 
 See the [forum](https://forum.joulescope.com/t/joulescope-ui-1-0-36-flagged-by-anti-virus-software/673) for the discussion.  If you check the antivirus history, you should be restore this file from quarantine.  Microsoft's Windows Defender is now fixed, so update your antivirus definitions.

chriszylstra, 2023-11-09:
Good call @mliberty1 this was the issue I faced. Thanks for the feedback. I read the forum post you linked.

We are using SentinelOne virus scanner and it got flagged as positive threat. 
[image]

All fixed now though, i'm not sure if you would like to submit SentinelOne for false postive analysis as well, but thank you for the support. 

-cz.

closed 2023-11-09 by chriszylstra

mliberty1, 2023-11-09:
We are already working on an alternate way to distribute the Joulescope UI using Nuitka.  The results are already much better according to [VirusTotal](https://www.virustotal.com/gui/file/23d15e82dffe60fedd021542980eac4e272e401ee4f829f2f1e113404d58c50a?nocache=1).

Since we will be changing this drastically, no need to submit the existing exe to SentinalOne.