jetperch/pyjoulescope_ui #248: Lock expressed unit regardless of the data range.

remy-luisant, 2024-01-11:
### UI version

1.0.48

### Your idea

Thank you for the wonderful device. We are finding it very useful.

At the company we often internally talk about current in units of uA. Sadly, in most ranges that we measure the label and scale on the left side will read a value in mA, since that is what is the best fit for the data.

It would be lovely to be able to lock the unit used for the scale to uA, even if it yields, for example, readings of 2200uA.

Thank you!

mliberty1, 2024-01-11:
Hi @remy-luisant - This enhancement makes sense to me for your use case and given value of 2200 µA.

However, I am sure we will get complaints if we implement it.  If someone locks it to nA and measures 1 A, then the widget as designed will not fit the full text for 1,000,000,000.  If the text is worth displaying, the UI must display all of it.

The existing UI has fixed widths for the y-axis tick labels, which works using SI prefixes.  By removing SI prefixes, this breaks.  Having variable-width y-axis tick labels would be very annoying, too, as they size up and down.

Having a **preferred** prefix may be possible.  For example, values up to 99,999 [prefix]A would display, but any larger range would ignore the preferred prefix and use the existing display method.

What do you think?

remy-luisant, 2024-01-11:
This would very much work for our case.

Thank you!

mliberty1, 2024-02-28:
Added in 1.0.58

closed 2024-02-28 by mliberty1