jetperch/pyjoulescope_ui #249: Bad readings on a zoomed-out waveform view.

remy-luisant, 2024-01-11:
### Joulescope model

JS220

### UI version

1.0.48

### What OS are you seeing the problem on?

Linux Ubuntu amd64 (PC)

### What happened?

The same data displays very differently depending on zoom levels. 

[image]
[image]

Note the difference in spike heights. This makes it all but impossible to hunt for the largest spike, since the heights are falsified when viewed from afar.

### What was expected?

The spikes to retain the sizes regardless of the zoom level.

### How to reproduce?

1. Record data with a spiky current use profile.
2. Zoom in.
3. Zoom out.

### Extra information

The GPU used is Intel Arc A770.

mliberty1, 2024-01-11:
Hi @remy-luisant - I think that you are seeing the as-designed behavior.  It looks like you have set **Min/Max** to **off**.  The purpose of **Min/Max** is exactly what you want; it displays the peaks regardless of zoom level.  I recommend that you set it back to **fill 2** to get what you want.

The **average** current line is just that, the average.  Each displayed pixel may represent many, many samples.  If you have a brief pulse and zoom in, you will see the exact pulse samples.  As you zoom out, then that pulse becomes averaged with the values beside it, which reduces the pulse amplitude.  This effect is the same as low-pass filtering the data.  There is no way to avoid this and have a sensible **average** current display.

Does this make sense?  Does setting **Min/Max** to **fill 2** give you the information you want?

remy-luisant, 2024-01-11:
It does make sense, however it does not quite resolve the matter.

The fill settings, as well as the line setting to a much lesser degree, add visual noise. We were hoping to be able to use the charts for a report, and clean output matters in this case. Furthermore, the averaging controls the odd current spike we have very well, letting the ranges stay lower than they otherwise would have been. 

A related problem is believability in the results. Were someone from the non-technical management see the following chart: 

[image]

And then moved it half of a screen to the left to find this, note the sizes of the two spikes on the left side:

[image]

Well... I think we would have some explaining to do as to why we are using this device. Even if there is a perfectly good reason for it, it would raise questions.

A possible workaround is a max mode, where the thick line follows just the maxima, possibly with a smoothing filter applied. 

In any case, I maintain that mere scrolling should not change the shape of the graph regardless of the settings chosen.

mliberty1, 2024-01-11:
Hi @remy-luisant - Unfortunately, you are using the wrong measurement to convey your point.  This plot is zoomed too far out with line widths that are thicker than the pulses, so pulse heights become meaningless.  The integral is always meaningful.  To compare two "equivalent" captures, you want to compare energy or charge.  Most customers use dual markers for this.

The average curve will always be the average.  Splitting "energy" over bins comes up in many places.  If the pulse is entirely contained within one pixel, then that pixel will be a larger value than if that same pulse spans two pixels.  The charge/energy is the same, in both, but the visual look is different.  We could enforce that scrolling and zooming is always quantized to keep curve shape.  However, this doesn't really help as other similar pulses in the same capture or other capture will have different "binning".  

One way around this is to reduce the sampling rate SIGNIFICANTLY so that these pulses are always spread across multiple pixels at the desired time zoom level.  This approach trades off temporal resolution for average peak consistency by always spreading the peak energy over multiple pixels.

remy-luisant, 2024-01-11:
I'm really sorry. I hate to be a pest, but the integral is of no use to us. Maybe more details will help.

We are using the Joulescope primarily as a debugging instrument, looking for a very rare rogue sensor activation. The board is unfriendly towards debugging and current is probably the easiest way to try to localize the problem in time. A board respin is months away at best. We are looking for a large current spike, while also having a spiky profile from intended activations of other sensors, as well as erratic extremely brief spikes in current that are of no interest to us. We need to cover a lot of time with our recordings, as well as be able to have a close look later on, to confirm if the spike is indeed the rogue sensor or not. This requires a fairly high sample rate, though we might be able to reduce.

The dual markers or energy measurements would not help here. The rogue sensor would come and go quickly, but would be of a different magnitude than the other sensor that is behaving correctly. This is how we can tell that the event we are looking for happened, by a different size. This is hard when the magnitudes of our spikes visually change. We can't feasibly zoom in on each spike from an overnight recording to see if it is the one we are after. Skimming from afar is the way to go, but that is not behaving very well.

My comment about needing to explain a fluidly varying shape to non-technical people in charge still stands, and same goes for suggesting a mode which displays primarily maxima instead of the averages.

We got it worked around by setting the trace opacity to 0 and the min_max_fill_alpha to maximum. It would be nice to have a similar option available without resorting to manually changing the opacities. The trace is now visually stable and this sufficiently closes the matter for us. Smoothing/post-processing would be welcome, but we will make due.

Thank you for your time! :D

closed 2024-01-11 by remy-luisant

mliberty1, 2024-01-12:
Great to hear that setting min_max_fill_alpha to 0xFF gives meaningful information for your case.

I configured the EVK1 to generate a sequence of pulses.  In the micropython terminal:

```
evk.stop()
evk.power_ldo(1.0)
evk.c_10U.on()
seq = [[100, 0x02], [2000, 0x80]]
evk.pio_start_sequence(10_000, seq, use_mask=True)
```

This should give the pulses with the same duration and current consumption.  Here is what I see for one zoom level with Min/Max turned off:
[image]

Note the varying pulse heights.

When you zoom in far enough so that each pulse is more than a pixel, you see the same amplitude:
[image]

One way to reduce the binning effect is to run a rectangular window (average) over the waveform.  Let's try length 3.  At joulescope_ui/widgets/waveform/waveform_widget.py line 1085 inside `on_callback_response`, insert this:
```python
            # length three rectangular filter
            d_avg_k = y[:, 0] * (1 / 3)
            d_avg = d_avg_k.copy()
            d_avg[1:] += d_avg_k[0:-1]
            d_avg[0:-1] += d_avg_k[1:]
            d_avg[0] += d_avg_k[0]
            d_avg[-1] += d_avg_k[-1]
            y[:, 0] = d_avg
```

Here is the full diff:

```text
--- a/joulescope_ui/widgets/waveform/waveform_widget.py
+++ b/joulescope_ui/widgets/waveform/waveform_widget.py
@@ -1082,6 +1082,16 @@ class WaveformWidget(QtWidgets.QWidget):
         elif response_type == 'summary':
             # self._log.info(f'response summary {length}')
             y = value['data']
+
+            # length three rectangular filter
+            d_avg_k = y[:, 0] * (1 / 3)
+            d_avg = d_avg_k.copy()
+            d_avg[1:] += d_avg_k[0:-1]
+            d_avg[0:-1] += d_avg_k[1:]
+            d_avg[0] += d_avg_k[0]
+            d_avg[-1] += d_avg_k[-1]
+            y[:, 0] = d_avg
+
             data = {
                 'x': x,
                 'avg': y[:, 0],
```

This "fixes" the pulse height for these short pulses:
[image]

However, it does cause autoscaling as new pulses arrive.  You may want to leave out the lines:

```
            d_avg[0] += d_avg_k[0]
            d_avg[-1] += d_avg_k[-1]
```

which causes this.  However, then the summary results displayed at the right will be different from when this filter is and is not applied.

You can run from source today with this edit to produce the graphs you want.

Neither the graph scaling nor the different summary results are desirable.  Unfortunately, this solution is suitable for merging into the UI at this time.