jetperch/pyjoulescope_ui #262: Waveform not working with two Joulescopes

foofel, 2024-03-11:
### Joulescope model

JS220

### UI version

1.0.58

### What OS are you seeing the problem on?

Linux Ubuntu amd64 (PC), Other (describe below)

### What happened?

Wen i connect a second Joulescope i can see it unter settings/devices and also read it with e.g. multiple multimeter views. But when using the Waveform i only see the options "default", "off" and "off" while the joulescope is actually running. 
[image]

### What was expected?

Show two joulescopes in the waveform widget.

### How to reproduce?

Connect two joulescopes and try to get them both to show in the waveform widget.

mliberty1, 2024-03-11:
Hi @foofel - In order to select the instrument for each trace, you have to right-click on the Waveform source, like this:

[image]

Note that left-clicking selects the active source, which is used for the right-hand side statistics and marker statistics.

Does this work for you?

foofel, 2024-03-11:
Awesome, thanks!

closed 2024-03-11 by mliberty1