jetperch/pyjoulescope_ui #267: possible problem with window size on translations

atsju, 2024-05-02:
### Joulescope model

JS220

### UI version

other

### What OS are you seeing the problem on?

Windows 11

### What happened?

Version 1.1.6 french win11

[image]

version 1.0.58 in english on win10

[image]

### What was expected?

Correct size for widget adapted or adaptable to text size.
Also how do you select language ? I have been unable to switch from French to English on new computer

### How to reproduce?

1. ...
2. ...
3. ...

mliberty1, 2024-05-02:
The **Flyout** widget in 1.1.6 and earlier has a fixed width.  It was really not wide enough for **Settings** before, and it is certainly not wide enough for French!

You can configure your computer to use English:

By default, the UI will use the OS language configuration. However, you can set the
LANG_JOULESCOPE_UI environment variable, which will override the OS language. Use the 2-letter
language codes above. At this time, the UI only offers basic language translations without country
codes.

In Windows, search for **Edit the system environment variables**.  Click **Environment Variables...".  Click New and create **LANG_JOULESCOPE_UI** set to **en**.

mliberty1, 2024-05-11:
Fixed in 1.1.7 which is available as alpha today.  You can now very carefully move the mouse to the right-hand-side of the flyout window.  The cursor will change to ↔.  You will then be able to click and drag to the desired size.  Note that the **Settings** widget in the flyout uses a separate size from everything else.  You will need to resize it separately.

closed 2024-05-11 by mliberty1