jetperch/pyjoulescope_ui #273: Add language selection method

mliberty1, 2024-06-03:
### Joulescope model

JS220

### UI version

1.1.9

### Your idea

The Joulescope UI introduced language translations in 1.1.  The language defaults to the system language.  The user can select a different language by setting the `LANG_JOULESCOPE_UI`.

We have feedback from a few customers who have the OS language localized but would like to run the Joulescope UI in English.  While this is possible with the environment variable, it requires extra steps.  The process is documented in the UI User's Guide, but no one reads that ;)

It would be better if the UI had a dialog or some approach to select the language.  It could display when the UI first launches or on request.

It would be ideal if the language could change dynamically.  Unfortunately, without a **LOT** of work, the Joulescope UI has to restart to change the language.

madmatt55, 2024-06-21:
Language change version is absolutely mandatory. With the current state of e.g. the german translation, I would suggest defaulting to english, it is in very bad shape and does not represent professional software quality.

mliberty1, 2024-06-21:
It sounds like the DeepL AI translations for German were particularly bad.  We have already made some changes based on feedback that will improve our future UI releases.  

In the meantime, you can change the UI language.  Set the system environment variable `LANG_JOULESCOPE_UI` to `en`.  In Windows, you can type "Edit environment variables" into the search box to bring up the control panel for this.  For more details, see Section 8 of the [Joulescope UI User's Guide](https://download.joulescope.com/products/joulescope_ui/users_guide/index.html).

atsju, 2024-06-21:
@madmatt55 I'm having a look at German translation.  https://github.com/jetperch/pyjoulescope_ui/pull/275
It's really long to do it all. I had 1 hour for  50lines only. ChatGPT seems to do well wen dirrected.
Please have a quick review. I'm not native speaker so it's quite difficult even with a good level. Please confirm following of my pain points:

- Zielgeräte VS Geräte ?
- How do you say sample ? Abtastzeitpunkt or Abtastpunkt or Messdaten ?

gkufner1, 2024-07-09:
Zielgeräte VS Geräte ? 
both is possible and will be understood. Zielgeräte usually means something like "devices under test"

How do you say sample ? Abtastzeitpunkt or Abtastpunkt or Messdaten ? 
For one sample I would say "Messpunkt", since it contains time and value.

atsju, 2024-07-09:
Thank you @gkufner1 

Not sure if I will have time to translate more any soon.

closed 2024-09-21 by mliberty1