jetperch/pyjoulescope_ui #276: Waveform trace color adjustment applies to all traces

StrandmonYellow, 2024-06-25:
### Joulescope model

JS220

### UI version

other

### What OS are you seeing the problem on?

Windows 11

### What happened?

I am running UI Version 1.1.10. 

When i want to change the trace color of one of the waveform channels, it applies to all of the traces. 

[image]

Then when i click on waveform.trace1, and change the color to red, it applies to all of the waveforms:

[image]

UI:
[image]

Color settings panel:
[image]

Also, the colors in the settings panel for waveform.trace2, waveform.trace3 and waveform.trace4 are not reflected in the UI. 

### What was expected?

I expect the naming to be the same. Trace0 - Trace3. When i change the color of waveform.trace1 i expect it to change the color of only this waveform in the plot. 

Additionally i expect there to be a setting for all of the channels (i, v, p, r, 0, 1, 2, 3, T)

I expect it to only change the color of trace 1 (Which is actually trace 0 as referred to in the settings and buttons in the UI. The settings says trace1 - trace4, and the UI shows 0-3 which is confusing).

### How to reproduce?

1. Start the application on Windows 11.
2. Do a measurement with multiple channels activated.
3. Settings are as shown below:

[image]

4. Stop the measurement
5. Right-click on one of the channels, and go to settings -> Colors tab.

mliberty1, 2024-06-25:
Hi @StrandmonYellow and thank you for the detailed issue writeup!  Let me first explain how things work today, and then see what makes sense given how you expect things to work.

The Joulescope UI **Waveform** widget displays multiple quantities (i, v, p, r, 0, 1, 2, 3, T).  We call each of these a **plot** within the **Waveform** widget.  Today, the Joulescope UI supports multiple simultaneous Joulescopes.  The Joulescope produces a waveform for each plot.  We call this a **Trace**.  When you change the color for trace1, you are changing the color for each trace in each plot for that source Joulescope.  You don't see this since you only have a single Joulescope connected, but it looks like this:
[image]

JS220-002122 is selected as active, and displayed on top in **trace1** color, which is yellow by default.   JS220-002557 is displayed in **trace2** color, which is cyan by default.  When you change **trace1** color, it changes the trace color for that Joulescope on all plots as you already observed, like this:
[image]

In other words trace1, trace2, trace3, and trace4 are linked to connected Joulescopes, not the general-purpose input quantities 0, 1, 2, 3 that are plotted in the **Waveform** widget.
---

It sounds like you want to change colors for a single plot.  As you can see, this would only work if a single Joulescope is connected.  The UI would need to ignore this setting if multiple Joulescopes are connected.  I can see why this would be desirable so that you can more easily find the signal of interest at a quick glance.

The next challenge is figuring out how to make this work.  Many of our plot-specific configurations do not show up in settings.  You adjust them by right-clicking on the plot.  We could add a **Color** item to the right-click menu.  It would have two options: **Trace** and **Clear**.  If you select **Trace**, it will bring up the color selector widget and adjust the trace color.  If you select **Clear** it will revert to defaults.  The UI would not display the **Color** item when multiple Joulescopes are connected and ignore any configured overrides.

What do you think?

StrandmonYellow, 2024-06-25:
Hi mliberty1,

This makes sense. Thankyou for the detailed explanation. I did search for something about this in the manual for the application, but could not find what I was looking for. I was not aware that the UI could be used with more than 1 joulescope simultaneously. 

As you said indeed, I would like this functionality because it make an obvious distinction between different plots within a waveform widget. 

Your suggestion sounds perfect! Do you need anything else from my side?