jetperch/pyjoulescope_ui #277: Changing names of channels is not being saved when exporting data

StrandmonYellow, 2024-06-25:
### Joulescope model

JS220

### UI version

other

### What OS are you seeing the problem on?

Windows 11

### What happened?

I am using UI version 1.1.10. 

When a measurement has been made, the names of the channels can be changed to anything. When I change the name, export the data and then load the file again, the names are back to default. 

When I save a file it shows:
[image]

When I reopen the file it shows:
[image]

### What was expected?

I expect the names to be saved when I reload my measurement. 

### How to reproduce?

1. Start the application on windows 11.
2. Perform a measurement with external GPIO as input.
3. Change the names by Right-clicking and entering a new name.
4. Export the data to a .jls file.
5. Load the file again, now the names are back to default. (0, 1, 2)

mliberty1, 2024-06-25:
Hi @StrandmonYellow - I can definitely understand that you would want the saved JLS to display the same in the **Waveform** widget as when you saved it.  

The original feature to label the plots is #255.  When you change a plot name, you are not changing a signal name.  Those exist internally, but the UI needs them in a certain way to display the correct information.  The plot name is only local to the **Waveform** widget.  

When you save a JLS file, you are saving each signal for each connected Joulescope.  JLS record is actually independent of any **Waveform** widget.  The plot names are only specific to a single **Waveform** widget.  You can actually add multiple **Waveform** widgets to the same view.  So, if you press the **Signal sample recording** button on the sidebar, and the UI knows that you have set plot names, the **Configure signal recording** widget could prompt you for signal names as **default** or any of the visible **Waveform** widgets.

Now, if you export a JLS from a **Waveform** widget, then it should automatically use the source **Waveform** widget for the plot names.

What do you think?

---

Developer notes: The JLS file signal names must not change.  However, we can save the plot names as arbitrary metadata on signal 0.  The existing code stores "notes" this way.  We could add a JSON structure that contains all of the Waveform plot overrides, like this:

```
{
  "id": "joulescope.ui.waveform.widget",
  "version": "1.0.0",
  "data": {
    "plots": {
      "i": {
        "range": ...
        "scale": ...
        "preferred_prefix": ....
        "name": ....
        "color": ....
      }, 
      "v": {
        ....
      },
     ....
   }
  }
}

StrandmonYellow, 2024-06-25:
Yes, this could be a valid solution to save the names without the need to save them in the notes of a recording.

closed 2025-03-04 by mliberty1

StrandmonYellow, 2025-03-04:
Great!!!