jetperch/pyjoulescope_ui #290: Boxcar downsampling option

Insulet, 2024-10-08:
### Joulescope model

JS220

### UI version

1.1.12

### Your idea

Create a native, UI selectable, downsampling option of boxcar filtering with no overlap.
This will eliminate artificial time domain over/undershoots.
The current sinc filter has great frequency domain fidelity, however I care more about the time domain:
What is my highest/lowest current? 
I understand that lower sample rates will mute out peaks. 
False peaks created from FIR filters with negative coeffs become edge glitches that trip up my analysis tools.

I do care about my DC accuracy, which is maintained with either method.

Our company just purchased 20 channels of this instrument solely because of this feature:
Keysight N6705C DC Power Analyzer User’s guide
https://www.keysight.com/us/en/assets/9018-01649/user-manuals/9018-01649.pdf
Page 165

mliberty1, 2024-10-09:
Hi @Insulet, and thanks for the recommendation.  A boxcar downsampling filter is super simple: for every Nth sample, output the average of the previous N samples.  You can see how the current code works in [downsample.c](https://github.com/jetperch/joulescope_driver/blob/main/src/downsample.c).  

Changing this to boxcar would only take me a few minutes.  Changing be configurable from the UI is more time, but still not more than a day.  I could have saved your company a ton of money if this is the only reason you went with the N6705C over the JS220!

We are actually working on something even better - adding boxcar filtering on-instrument so that you also reduce USB bandwidth.  This will allows for even more JS220s on the same USB root hub.

If you are allowed to share, what output sample rate do you normally use?

Insulet, 2024-10-09:
When do you anticipate the boxcar down-sample release?

Output sample rate. I try to run as fast as possible. This can make for giant data files.
N6705C data logger min sample period is 0.1024mS*things sampled.
In the worst case this is 4channels of V and I, so a period of 0.8192mS.

>1mS different islands of current are hard to separate as BLE spikes are muted.

[External email] Spot the Sham, Slam the Scam.

mliberty1, 2024-11-08:
Fixed.  JS220 FPGA 1.3.0 implements on-instrument sinc1 (boxcar) filtering.  Not only does this on-instrument filtering address this issue, it also reduces USB throughput allowing more JS220's to run on the same USB root hub.

To get full support, you need FW 1.3.0, joulescope_driver 1.7.0 and UI 1.2.0.  We are starting testing today.  Assuming testing goes well, we should have a release within two weeks.

closed 2024-11-08 by mliberty1