jetperch/pyjoulescope_ui #293: Min/max waveform artefact on 2K display

AloyseTech, 2024-11-29:
### Joulescope model

JS220

### UI version

1.2.2

### What OS are you seeing the problem on?

Windows 11

### What happened?

Waveform min/max traces have artefacts and some lines of the drawn polygon seem to converge on the waveform window center.

[image]

### What was expected?

Normal min/max waveform

### How to reproduce?

1. Open UI window on a 2K display
2. Start acquisition with `show_min_max` setting on `fill1` or `fill2`

### Extra information

If I resize the window while the acquisition is running, the artefacts only appear above a certain width. If the window is resized below ~2300px they disappear.

mliberty1, 2024-11-29:
Can you share your monitor resolution and graphics card?

My normal machine is 3840 x 2160.  However, this is always less than 2300 px.

mliberty1, 2024-11-29:
We have the same monitors!
[image]

I usually run at full res 3840 x 2160 with no issues.

I am guessing that this is an Intel graphics OpenGL bug.  Is that the most recent Intel graphics driver for your machine?

AloyseTech, 2024-12-04:
All drivers are up to date. I will try this and report back.

AloyseTech, 2024-12-11:
Disabling **opengl** in the settings solves the issue (but slows down the UI a lot obviously).

Re-enabling opengl and setting the rendered to **software** does not solve the problem. Setting it to **desktop** do solve the problem though. For some reason I think it was on **software** by default. When setting it to **desktop** I had the popup about https://github.com/jetperch/pyjoulescope_ui/issues/216 but no problem so far. And the UI is much more responsive.

mliberty1, 2024-12-11:
Great to hear you have a working solution!

The dodgy Intel integrated graphics support for OpenGL has been an issue.  We added those workarounds a while ago.  Maybe Intel has actually fixed things?  If you do run into #216, please post again!

alex-sony, 2025-02-17:
I have the same issue where increasing the width of the windows the signals in the waveform disappear. I tried disabling opengl but the issue is still there. I have a 5120px width resolution and it starts happening when the width of the Joulescope UI window goes over ~4343px. 

Monitor resolution : 5120 x 2160p
Graphic card : Nvidia RTX4090

[image]

mliberty1, 2025-03-04:
[joulescope_driver 1.7.4](https://github.com/jetperch/joulescope_driver/releases/tag/v1.7.4) increases `JSDRV_STREAM_DATA_SIZE` from 64 kB to 256 kB to fix the truncation issue reported by @alex-sony .

The OpenGL software backend (Mesa3d) still has a problem with large polygons.  The painter `drawPolygon` works correctly with Qt rendering and hardware OpenGL rendering, so it is not a logical error in the UI.  The best workaround is to select min/max `lines` if you need software rendering on higher-resolution screens.

commit a7f94139 (2025-03-04): Updated pyjoulescope_driver from 1.7.2 to 1.7.4, fixes #293