jetperch/pyjoulescope_ui #305: Annotations jumping around in waveform widget

fridfinnur-staroddi, 2025-03-17:
### Joulescope model

JS220

### UI version

1.2.6

### What OS are you seeing the problem on?

Windows 11

### What happened?

While adding dual vertical markers to a streaming capture (while also recording), the waveform occasionally jumps around with respect to the annotations. The vertical annotation mode was set to absolute.

### What was expected?

The annotations should stay at fixed locations relative to the waveform, as they did in 1.2.5.

### How to reproduce?

1. Start streaming samples and recording. (100khz)
2. Add a dual vertical marker annotation.
3. Wait approximately 10 seconds.

### Extra information

Screen recording attached.

P.S. Have been loving the JS220 so far, amazing instrument!

https://github.com/user-attachments/assets/6aad3769-4ca4-416d-942f-9f301f4e1734

fridfinnur-staroddi, 2025-03-18:
Hi Matt, thank you for the prompt reply!

Yes, I see now that my PC's clock is 1.1 seconds off so that's probably the source of the problem.
I updated the firmware just a few days ago: 

[image]

Even then, I have been using annotations to do charge measurements on pulses that are only a few milliseconds long, so them jumping around by a few milliseconds is still not ideal!
Is there any way to lock the annotations to sample time, or will I just have to finish recording before annotating?

mliberty1, 2025-03-18:
Hi @fridfinnur-staroddi - Thanks for the `Device Information` image.  I agree you are up to date.

We have been working to improve time synchronization, implementing multiple enhancements over the past couple of years.  The good news is that this behavior should have improved with the newer Joulescope UI releases.  I suspect that something else is happening, which causes you to notice it more now.  Longer duration RAM buffers definitely accentuate the issue.  Improving the host time accuracy should also help.  

Here is a script that implements the [Microsoft recommendations](https://learn.microsoft.com/en-us/windows-server/networking/windows-time-service/configuring-systems-for-high-accuracy).  Start PowerShell or Command Prompt as administrator, and then paste these commands:

```
sc config W32Time start=auto
sc start W32Time

reg add HKLM\SYSTEM\CurrentControlSet\Services\W32Time\Config /v MinPollInterval /t REG_DWORD /d 6 /f
reg add HKLM\SYSTEM\CurrentControlSet\Services\W32Time\Config /v MaxPollInterval /t REG_DWORD /d 6 /f
reg add HKLM\SYSTEM\CurrentControlSet\Services\W32Time\Config /v UpdateInterval /t REG_DWORD /d 100 /f
reg add HKLM\SYSTEM\CurrentControlSet\Services\W32Time\Config /v FrequencyCorrectRate /t REG_DWORD /d 2 /f
reg add HKLM\SYSTEM\CurrentControlSet\Services\W32Time\TimeProviders\NtpClient /v SpecialPollInterval /t REG_DWORD /d 64 /f
w32tm /config /update
net stop w32time && net start w32time
w32tm /resync
```

We have multiple motivations for improving x-axis annotation stability while streaming:
1. **Trigger** widget needs it to accurately add markers
2. Upcoming **Serial Console** widget needs to add single markers and text annotations.
3. You want it to annotate while capturing long durations

Synchronizing time is an old and challenging problem.  We have historical reasons for how things currently work, which is only to use the most recent time map update.  See [`utc_to_samples`](https://github.com/jetperch/joulescope_driver/blob/90ac2cb02e5f981e545c5d7b99dd5ef40510e3ff/src/buffer_signal.c#L143).  

One possible fix is to modify `utc_to_samples` to use the closest, previous time_map update.  One known complication is then JLS export, which would also need to export all of these in addition to the samples.

Interestingly, the JLS file format already uses the [closest time entries](https://github.com/jetperch/jls/blob/eb10489503c37f9bd440e1082d00e75c90de58ab/src/tmap.c#L165) to convert from UTC time to sample time.  I do not think that this difference matters in any way for the current UI other than a JLS recording made from the UI and a JLS export of the waveform may be very slightly different.  However, all charge & energy integration is performed based on sample-time at 25 ppm accuradcy, so the integrated values would be the same.

The JS220 FPGA 1.3.2 finally has time synchronization exceeds the JS220's ±25 ppm oscillator accuracy, even over a 1 minute duration.  The JS220 is accurate to 25e-6 * 60 s / minute = 1.5 ms / minute.  The JS220 converges to the host computer time better than 1.5 ms accuracy within seconds.  While we previously had reasons to update historical times, we can probably now simply use the time initially given to an incoming JS220 sample.  The big wildcard is host computer time accuracy, which directly adds error.

We will continue to investigate.

fridfinnur-staroddi, 2025-03-18:
Thank you for the informative write-up Matt, I can see that it is indeed a tricky problem!

I configured the w32time service as per your recommendations (and pointed it at my country's ntp pool), and time.is now reports no significant deviation. To characterize the drift I ran a 15 minute recording where I added a single marker to the start of a pulse each time it jumped, pictured below:

[image]
Over 15 minutes the pulse shifted forward by 32.5ms, mostly in small jumps forward, but occasionally in large jumps and twice in small jumps backwards.

Thank you for mentioning the Trigger widget, it turns out it can automatically add the annotations that I've been adding by hand, but much quicker and more consistently!
The only downside is that if it annotates a 50ms pulse at the start of a 20 minute recording, the cursors will end up shifting by enough to totally miss the pulse by the end of it.
Being able to confidently use the Trigger widget would be a great productivity boost, I have been meaning to write a script to do pretty much exactly what it does on recorded files (that is, add dual markers between rising/falling edges).

closed 2025-03-27 by mliberty1

mliberty1, 2025-03-27:
This issue is fixed by pyjoulescope_driver 1.8.0 and Joulescope UI 1.3.0.  While I expect this version to eventually be released as stable after we complete testing, you can [download as alpha](https://download.joulescope.com/joulescope_install/index.html) today.

mliberty1, 2025-03-28:
Well, 1.3.0 failed our release testing.  We completely broke JLS file open.  1.3.1 in progress...