jetperch/pyjoulescope_ui #309: <title>Joulescope views with multiple joulescope

alexandre200, 2025-04-18:
### Joulescope model

JS220

### UI version

1.3.1

### Your idea

Enhancement ideas:
Allow for multipe voltage, current or power monitoring accors multiple joulescope (Not stuck with viewing of one channel at a time), for instance, power of channel rail 1 through joulescope 1 with power of rail two through joulescope 2 seen in parrallel (option to add them ?)

Allow for joulescope channel renamin, could really help when working with varying voltage for instance, I had to unplug replug joulescope to understand wich one was wich when measuring.

alexandre200, 2025-04-18:
Hello [mliberty1](https://github.com/mliberty1), I must say that I did not found the features I was looking for, but indeed they are here. Apologize for the confusion, what led me to not understand how the channel selection works is the fact that you have to right click instead of left click for enable and selecting the channel, therefore I was not understanding the channel concept properly, since I was left with a single channel.

Regarding the possibility to do math on the signals, I understand the complexity in selecting a decimation or interpolating to mitigate this differents joulescope clocks. Was a nice to have, maybe can it be done only for the avg, min, max, rms values ?
For instance here : 

[image]

If this is allready a feature and I missed it then sorry in advance.

Thank you for your quick responses and wish you a good day 👍

mliberty1, 2025-04-18:
HI @alexandre200 - As of Apr 2025, the **Waveform** widget displays the statistics (both right side, single vertical marker and dual vertical marker) for the selected channel.  You select a channel by left-clicking on it, and it is indicated with a rounded rectangle, like this:

[image]

As you found, you right-click on a channel to select the source.

So, I think the possible features are:
1. Display the statistics as a math function of any combination of channels when selected.  No change to signal plots.
2. Display a math channel that is the math function of any combination of sources, even if not configured as channels.  Interpolate to align signals across sources as needed.  Display statistics on the math channel when selected.