jetperch/pyjoulescope_ui #315: SEGFAULT

zfields, 2025-05-13:
### Joulescope model

JS220

### UI version

1.3.3

### What OS are you seeing the problem on?

Linux Ubuntu amd64 (PC)

### What happened?

```$ joulescope
*****************************
* Joulescope User Interface *
* UI Version = 1.3.4        *
*****************************

info = {
  "joulescope": {
    "ui_version": "1.3.4"
  },
  "platform": {
    "name": "linux",
    "python_version": "3.12.3 (main, Feb  4 2025, 14:48:35) [GCC 13.3.0]",
    "platform": "Linux-6.8.0-59-generic-x86_64-with-glibc2.39",
    "processor": "x86_64",
    "executable": "/usr/bin/python3",
    "is_release": false
  }
}

=====
WARNING:2025-05-13 10:03:47,051:backend.c:380:jsdrv:zero length bulk in transfer
WARNING:2025-05-13 10:04:06,464:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:06,473:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:06,473:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:06,473:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:06,474:jsdrv.c:397:jsdrv:u/js220/001004/s/i/ctrl but device already removed
WARNING:2025-05-13 10:04:06,475:jsdrv.c:397:jsdrv:u/js220/001004/s/v/ctrl but device already removed
WARNING:2025-05-13 10:04:06,475:jsdrv.c:397:jsdrv:u/js220/001004/s/p/ctrl but device already removed
WARNING:2025-05-13 10:04:06,475:jsdrv.c:397:jsdrv:u/js220/001004/s/stats/ctrl but device already removed
WARNING:2025-05-13 10:04:06,994:js220_usb.c:541:jsdrv:d_ctrl_req(2) returned 8
WARNING:2025-05-13 10:04:06,995:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:07,512:pubsub.py:902:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-001004-UPDATER/actions/!finalize
WARNING:2025-05-13 10:04:08,534:js220_usb.c:541:jsdrv:d_ctrl_req(2) returned 8
WARNING:2025-05-13 10:04:09,048:pubsub.py:902:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-001004-UPDATER/actions/!finalize
WARNING:2025-05-13 10:04:10,130:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:10,130:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:10,130:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:10,130:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:10,131:js220_usb.c:541:jsdrv:d_ctrl_req(2) returned 8
WARNING:2025-05-13 10:04:10,584:pubsub.py:902:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-001004-UPDATER/actions/!finalize
WARNING:2025-05-13 10:04:10,749:backend.c:380:jsdrv:zero length bulk in transfer
WARNING:2025-05-13 10:04:10,750:js220_updater.py:214:joulescope_ui.devices.jsdrv.device.u.js220.001004:Abnormal condition state=5, target=app, restart
WARNING:2025-05-13 10:04:10,751:js220_usb.c:541:jsdrv:d_ctrl_req(2) returned 8
WARNING:2025-05-13 10:04:10,752:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:10,752:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:10,752:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:11,098:pubsub.py:902:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-001004-UPDATER/actions/!finalize
WARNING:2025-05-13 10:04:11,268:js220_usb.c:541:jsdrv:d_ctrl_req(2) returned 8
WARNING:2025-05-13 10:04:11,269:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:11,665:pubsub.py:902:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-001004-UPDATER/actions/!finalize
WARNING:2025-05-13 10:04:11,830:backend.c:380:jsdrv:zero length bulk in transfer
WARNING:2025-05-13 10:04:11,831:js220_updater.py:258:joulescope_ui.devices.jsdrv.device.u.js220.001004:updater2_version_check, but in app
WARNING:2025-05-13 10:04:11,832:js220_usb.c:541:jsdrv:d_ctrl_req(2) returned 8
WARNING:2025-05-13 10:04:11,832:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:11,833:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:12,205:pubsub.py:902:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-001004-UPDATER/actions/!finalize
WARNING:2025-05-13 10:04:13,267:js220_usb.c:541:jsdrv:d_ctrl_req(2) returned 8
WARNING:2025-05-13 10:04:13,267:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:13,656:pubsub.py:902:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-001004-UPDATER/actions/!finalize
WARNING:2025-05-13 10:04:18,647:js220_usb.c:541:jsdrv:d_ctrl_req(2) returned 8
WARNING:2025-05-13 10:04:18,647:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:18,647:backend.c:403:jsdrv:bulk_in error 1
WARNING:2025-05-13 10:04:19,032:pubsub.py:902:joulescope_ui.pubsub:Publish to unknown topic registry/JS220-001004-UPDATER/actions/!finalize
WARNING:2025-05-13 10:04:19,200:backend.c:380:jsdrv:zero length bulk in transfer
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Painter path exceeds +/-32767 pixels.
Segmentation fault (core dumped)```

### What was expected?

No SEGFAULT

### How to reproduce?

I recorded a very long sample > 20 minutes

mliberty1, 2025-05-13:
Hi @zfields - Well, it looks like the UI updated the JS220 firmware, and then started operating normally.  Those `Painter path exceeds +/-32767 pixels.` can't be good.  However, it's also not a lot to go on.

Could you submit an issue report from within the Joulesope UI?  It may have more information.  Start the Joulescope UI (if not already running) and then select **Help** → **Report Issue**.  It keeps the most recent 10 logs (I think), so hopefully this SEGFAULT will still be there...

mliberty1, 2025-05-14:
We received the issue report from @zfields .   Thank you @zfields !

It looks like the SEGFAULT occurred when exiting the Joulescope UI.  The 20250513_150346_269430 log shows this at the very end:

```
INFO:2025-05-13 12:12:44,149:buffer.c:412:jsdrv:signal remove 0
INFO:2025-05-13 12:12:44,149:buffer.c:412:jsdrv:signal remove 0
INFO:2025-05-13 12:12:44,149:buffer.c:205:jsdrv:buffer_alloc 5368709120
INFO:2025-05-13 12:12:44,149:buffer.c:229:jsdrv:0 B/s -> -2147483648 seconds
```

That last line looks bad!

The last line of clean exits should be something like:
```
INFO:2025-05-13 10:01:16,355:main.py:993:joulescope_ui.main:exit 0
```

mliberty1, 2026-04-27:
We have found and fixed several use-after-free cases for Linux, which will be included in the next release.  We are targeting the end of May or early June.