jetperch/pyjoulescope_ui #316: Degraded graph performance wi 1.3.4

diggit, 2025-05-16:
### Joulescope model

JS220

### UI version

other

### What OS are you seeing the problem on?

Other (describe below)

### What happened?

After updating from 1.3.3 to 1.3.4, the graphical performance dropped. I had framerate around 35 FPS. With 1.3.4 it is only ~6 FPS. I also noticed, **joulescope_ui now starts under xwayland**.

### What was expected?

Similar framerate of the graph in both versions.

### How to reproduce?

1. install 1.3.3 and 1.3.4
2. observer different maximum framerates of the graph

### Extra information

OS: Archlinux, kernel 6.14.5
Sway
Intel(R) Core(TM) i7-6700HQ with integrated graphics
Display via TB, 4096x2160 @ 60.000 Hz (+ 1 additional display)
Maximized joulescope_ui wuindow, almost fullscreen
framerate limit set to vsync

Let me know what to provide or try out.

mliberty1, 2025-05-16:
Unfortunately, Qt6 with the Advanced Docking System (the framework we use) has problems under Wayland.  Reduced frame rate is better than broken.  See https://github.com/jetperch/pyjoulescope_ui/issues/311.  

The Joulescope UI now [automatically detects Wayland](https://github.com/jetperch/pyjoulescope_ui/blob/0394096675e88be17dcdd3f0f11899db75fb8526/joulescope_ui/main.py#L877).  

I am not sure what other effects this has, but setting `XDG_SESSION_TYPE` before invoking the UI may work, like this:

```
XDG_SESSION_TYPE="none" ./joulescope_launcher
```

If you run from source, you can also edit this detection code to do whatever you want.  

Unfortunately, we have to wait until https://github.com/githubuser0xFFFF/Qt-Advanced-Docking-System/issues/714 gets fixed before allowing Wayland by default.

mliberty1, 2025-05-16:
Unfortunately, this ADS issue is not high enough priority for us to help fix, either.  It looks like your CPU is 10 years old.  More modern CPUs suffer from degraded frame rate, but not quite so drastically.

For the next release, we could add an environment variable check to explicitly bypass this.  You would still have the Wayland issues whenever you have an undocked widget.  Unfortunately, even right-clicking a widget and selecting settings causes an undocked widget.